Ultimaker/Cura #6118: [4.2.1] Supports cros the brim

Liger0, 2019-08-01:
**Application version**
4.1.2

**Platform**
Win10 x64

**Printer**
Custom dual extruder

**Reproduction steps**
Use tree supports and brim

**Actual results**
They cross each other creating blobs and eventual fail

**Expected results**
there should be no support where there is the brim

**Additional information**
[image]
[brim and support crossing.zip](https://github.com/Ultimaker/Cura/files/3458040/brim.and.support.crossing.zip)

Vandrasc, 2019-08-05:
Hi @Liger0 , is the support brim enabled? Then the support should indeed be generated above the brim, for better adhesion of the support.

Liger0, 2019-08-05:
Hi there. What you see in the preview is only the first layer, not the second.
On the first layer, different features cross each other, thus the nozzle clogs and the print fails, also the nozzle drags and moves out the first layer.
I never printed that model nor I intend to, I just use it because it is a simple yet complex enough model that helped me to test out settings and find multiple cura bugs until now.
The support brim is enabled, and the project is shared under the image.

Ellecross, 2019-08-07:
Will it help if you disable the Support Brim? 

If you do, it will have only the support in the brim and one line with extruder 1 at the outermost. 

[image]

Ghostkeeper, 2019-08-08:
I think the actual problem here is that you have both Tree Support and ordinary Generate Support enabled. This will generate the support twice which is not generally what you want and causes overextrusion.

We're planning to remove this possibility.

Vandrasc, 2019-08-09:
We created a ticket internally for removing the possibility of enabling both type support, devs for our reference see CURA-6711.

closed 2024-12-07 by github-actions[bot]