Ultimaker/Cura #6119: [4.2.1] Updating custom profile doesn't save settings

FloRad, 2019-08-01:
When updating a custom profile and then clicking the button "Update profile with current settings/overrides" the settings are not saved and instead a new profile titled "empty" is created. This new profile also seems to be corrupt sometimes, and does not take over all settings.

Vandrasc, 2019-08-05:
Hi @FloRad , what printer are you using? 

I am using the following reproduce steps:

1. Add UM3 and create a custom profile
2. Change the infill density to 50%
3. In the profile drop-down I select "Update profile.." --> my custom profile updates the new settings. 

Did I leave anything out? 

Thanks!

FloRad, 2019-08-05:
Hi @Vandrasc, sorry for not adding my printer. I am using a Creality Ender 5.
My uneducated guess is that the cause may be due to Creawesome.

Vandrasc, 2019-08-05:
Hi @FloRad , no worries, thank you for the quick reply. 
 Ah, yes, that could be. Your custom profile was created in a previous version then? In 4.2.1 if I try to reproduce the issue doesn't occur, only when upgrading from a version before the Creawesome.

FloRad, 2019-08-05:
Yes, my custom profile was created in Cura 4.0 and later carried over into 4.1 where it is still in use.

Fernando-Checa, 2019-08-06:
Same here.. just updated to ver. 4.2.1 and updated a couple of setting. When I was saving the file got corrupted.
Next time I loaded Cura, it detected a corrupted file and offered to reset all settings.. Not very nice.

onkytonk, 2019-08-06:
Using Cura 4.2.1 with an Ender 3 Pro and encountering the same issue. "Update profile with current settings" does not save the current settings. Many settings are different after saving, and then the profiles become corrupt, and Cura asks me to reset all my profiles, and remove my current printer profile.

Ghostkeeper, 2019-08-08:
I'm also unable to reproduce this using the Ender 5 or the Ender 3 Pro. These are my exact reproduce steps:

1. Add Creality3D Ender 5 printer.
2. Change Infill Density to 42%.
3. Save as a new profile called "Infill of Life".
4. Change Infill Density to 43%.
5. Update the profile with current changes (from the profile drop-down menu).

Are you changing a different setting? We might be looking at #6167 here, which can have very strange behaviour if the setting you're looking at is not actually the setting value that Cura remembers.

FloRad, 2019-08-08:
Hi! This happned when changing layer height, Retraction Speed and Retraction Distance. Please not that both the printer and the profile have existed before I upgraded to Cura 4.2.1, since the upgrade took over the settings from my Cura 4.1 Installation, so no printer was added.

onkytonk, 2019-08-13:
I created a custom profile completely from scratch in 4.2.1. Everything else I imported was getting corrupted. 
I added a model, and made a few tweaks to that profile. I selected "Create profile from current settings."

Settings I changed was:
Special modes: Spiralize (from off)
Fan: On at layer 4 (from 2)
Bed temp: Initial 55 degrees 
Bottom pattern initial layer: Concentric (from lines)

Settings that GOT changed:
Line width: 4.4 (from 4.0)
Top surface skin: 0 (from 2)
Top layers: 6 (from 4)
Bottom layers: 6 (from 4)
Z seam alignment: User specified (from sharpest corner)
infill line distance: 6.0mm (from 2.0mm)
infill pattern: Cubic (from gyroid)
infill overlap percentage: 30% (from 10%)

And who knows what else got changed? 

I can send you my profiles to try the same thing if you like?

Ghostkeeper, 2019-08-13:
I'm not able to reproduce that either. Here are my exact reproduce steps:

1. Add an Ender 3 printer. There is no definition for Ender 3 Pro that you claim to be using so this is the closest I could get.
2. Enable Spiralize.
3. Regular Fan Speed At Layer was already set to 4, so I didn't change that.
4. Build Plate Temperature was already set to 55, so I didn't change that.
5. Changed Bottom Pattern Initial Layer to Concentric.
6. Save the two changes as a custom profile.

Before the change, the line width was already 0.44mm and stays at 0.44mm. The Top Surface Skin Layers stays at 0. The Top Layers stays at 4, the Bottom Layers stays at 4, Z Seam Alignment stays at User Specified, Infill Line Distance stays at 6.6mm, infill pattern stays at Cubic and Infill Overlap Percentage stays at 30%.

So yes, I guess I do need those profiles from you in order to reproduce this.

Since we have seen some update problems with the setting values in the GUI, it might also be worth your time to restart Cura to see if it's that, before validating the setting values. The displayed number might not be what it actually slices with.

onkytonk, 2019-08-13:
Yes, I'm usingbthe ender 3 profile.

It keeps happening with this profile. Even after restarting Cura. 

I would suggest you try creating a custom profile from an existing custom profile. 
(I've atttached the one I use anyway)
[LH PLA 4.2.zip](https://github.com/Ultimaker/Cura/files/3499209/LH.PLA.4.2.zip)

I've also noticed the bed dimensions or something changing. The dark grey boarder on the bed gride gets larger when the new profile is created. 
[image]
[image]

onkytonk, 2019-08-14:
I also keep getting a profile called "empty" created.

Ghostkeeper, 2019-08-14:
> I've also noticed the bed dimensions or something changing. The dark grey boarder on the bed gride gets larger when the new profile is created.

From your screenshot, it looks like it's just doing that because your skirt distance is increasing due to the profile change. If the skirt is farther away, Cura will prevent your object from being placed too close to the edge so that the skirt isn't printed outside of the build volume.

onkytonk, 2019-08-14:
Then that setting is obviously being changed too.

Ghostkeeper, 2019-08-14:
I'm still unable to reproduce the issue, even with your profile there :confounded: 

Could you write down exactly what you do, step by step, to get at it? These have been my reproduce steps:

* Add an Ender 3 printer.
* Import your profile.
* Switch to your newly imported `LH PLA 4.2` profile.
* Enable Spiralize.
* Save the changes as a new profile called `bla`.
* Verify that the line width is still 0.4mm (as specified by your profile) instead of 0.44mm.

onkytonk, 2019-08-14:
That’s so strange. It keeps happening to me. 
As mentioned above, there were multiple settings I changed.
But what you’re saying is basically what I’m doing.

onkytonk, 2019-08-14:
The messed up profiles don't show any settings in profile management either:

[image]

onkytonk, 2019-08-14:
I just tried creating a new printer profile. I re-added an Ender 3. 
I changed a few settings in my custom profiles and saved as a new profile. It worked as it should have.
I had a hunch it may have had to do with using custom materials too, but that didn't affect it either. 
So still not sure what's causing my issues. 
At least for now, it seems to be ok with my new printer profile.

Ghostkeeper, 2019-08-15:
I think we might be running into this issue here: https://github.com/Ultimaker/Cura/issues/6167

onkytonk, 2019-09-23:
Was there ever a solution for this? Same problem exists in Cura 4.3 beta.

Ghostkeeper, 2019-09-24:
Yeah, that issue #6167 was previously blamed for the issue you were having, but since you're still experiencing it, it's probably not the cause then.

closed 2024-12-07 by github-actions[bot]