Ultimaker/Cura #6120: [4.2.1] Tree support z distance, number of interface layers.

Liger0, 2019-08-01:
**Application version**
4.2.1

**Platform**
Win10 x64

**Printer**
Custom dual extrusion

**Reproduction steps**
I set a z distance between on bottom and top both to 0 mm.
I use a 0.12 layer height, and as I want this profile to just work also if I use bigger layer heights, I want there to be at least 0.1mm of bottom interface layer and 0.2mm of top interface layer.

**Actual results**
There is a fair amount of z distance and the number of interface layers is random

**Expected results**
For a 0.12 LH I expect there to be 0 empty layers (z distance is 0!), 1 bottom interface layer (0.10 rounded to 0.12), and 2 top interface layers (0.2 rounded to 0.24).

**Additional information**
[tree support interface.zip](https://github.com/Ultimaker/Cura/files/3458208/tree.support.interface.zip)

AbeFM, 2019-08-01:
I've noticed the interface layers seems random sometimes. The first time I had that issue it was "interface horizontal expansion" (which can ONLY be set negative, preventing using it to encourage adhesion).

Also, if you can find or make a model that is a LOT less complicated for testing the slicing on, this would go quicker. see https://community.ultimaker.com/topic/29010-ultimaker-cura-42-released/ or https://github.com/Ultimaker/Cura/issues/6123 - you could just slice off everything over the coat tip and most of the ways above the legs.

I think your basic issues, other than the broad support issues I've been asking about for the better part of a couple of years, is that the area you're trying to support is very small. 

[RickCoatTip.stl.txt](https://github.com/Ultimaker/Cura/files/3458453/RickCoatTip.stl.txt)

I don't think I have a lot else to offer here. :-)

Liger0, 2019-08-01:
Using that part of the model only 2 little points are supported. It will 100% fail. That because supports don't work on gap filling and all the bottom layers are composed by gap filling extrusion lines.

Ghostkeeper, 2019-08-11:
This is a duplicate of https://github.com/Ultimaker/Cura/issues/5759.

closed 2019-08-11 by Ghostkeeper

nallath, 2020-07-10:
Fixed in 4.7