Ultimaker/Cura #6121: [4.2.1] Support can't reach the model

Liger0, 2019-08-01:
**Application version**
4.2.1

**Platform**
Win10 x64

**Printer**
Custom dual extruder

**Reproduction steps**
Use normal supports, no towers

**Actual results**
One piece of the support is far off from the point it should support (I know that supports never support the filled gaps features which usually leads to failed prints, but here it shouldn't be the reason)

**Expected results**
>The support should touch the model

**Additional information**
[support can't reach the body.zip](https://github.com/Ultimaker/Cura/files/3458268/support.can.t.reach.the.body.zip)
[image]

AbeFM, 2019-08-01:
I might print that on its back, or draw up manual supports for the coat bottom!

I first suspected minimum support interface area, but it didn't help. 

[image]
Setting the interface for Lines instead of ZigZag seems to have worked, even without the minimum area change described, I think the higher density pattern "saw" what's above it better.

Ghostkeeper, 2019-08-08:
The [support areas are generated](https://github.com/Ultimaker/CuraEngine/blob/master/docs/generating_areas.md#support) first, before it's looking at the actual pattern to fill them with.

The problem in this case is that the support area is too thin to fit a line there. If you set the support line width to 0.1mm, it builds the support fine (but it'd be too small to print with the 0.4mm nozzle size). That's also why the Lines pattern does generate support there: The zig-zag needs to start turning the corner half a line width away from the border in order to not go outside of the border, while the Lines pattern can just draw a 0.2mm line segment at 0.4mm line width.

We have several options to fix this. The most appropriate would be Use Towers since it's made to fix this type of problem but it won't work for you since it's a long continuous border that will have a significant area that needs to be supported, so it won't create a tower. A better option in your case would be to set Support Horizontal Expansion to 0.2mm or so, which will make the support areas wide enough to be printed by your printer.

The reason for the "floating" bit in the original bug report is that the X/Y distance removes a sliver of the support such that it's wide enough to print there on the lower layers but not wide enough in the upper layers. Rick's calf is coming too close to the support there.

Ghostkeeper, 2019-08-09:
This is not desired behaviour, but it is currently expected. Support Horizontal Expansion is a reasonable way to deal with this, so we don't need a different solution.

closed 2019-08-09 by Ghostkeeper

Liger0, 2019-08-09:
Is this the same reason why this model only has one side of the support reaching its base?
[CCR10S_CR-10_HD-modular-mount-V6_mount holder.zip](https://github.com/Ultimaker/Cura/files/3487700/CCR10S_CR-10_HD-modular-mount-V6_mount.holder.zip)

AbeFM, 2019-08-09:
Seems like a feature request, at this point - moreso than a bug.

Perhaps this is more "how do I print this" and should go on the forums until it's a "bug".

Set support distances more reasonable, and z-overrides-x/y.
[image]
There's still bugs with those features but it should fix your issue.

Liger0, 2019-08-09:
Well.. no one ever replies to me on the forum.