Ultimaker/Cura #6123: V3+ Minimum Support X/Y Distance Not Respected on Floors

AbeFM, 2019-08-01:
**Application version**
3.x - 4.2.1

**Platform**
win10/NA

**Printer**
Several custom multi-extrusion printers

**Reproduction steps**
Set Support XY Distance to  number (i.e. 1 mm), select Z Overrides CY, and set Minimum Support XY Distance to something smaller (i.e 0.2 mm).

**Actual results**
[image]

**Expected results**
I believe the Minimum XY Distance should be respected at the lower object/support interface as well as at the top. Perhaps using an Absolute Value for Z distance would work, I'm unsure and happy to discuss if I can get my point across about what it happening.

**Additional information**
My goal: To have my supports stick to my model without ruining side walls.
My approach: Print supports so their first layers touch the model.
My issue: CURA specifically avoids the model when printing support, ignoring Minimum XY Distance setting..

I would like to use the "Support XY Distance" to prevent my supports from merging with the object (embedding PVA into the print which leaves holes and rough surface on near-vertical and vertical walls), and use "Minimum Support XY Distance" to allow the support to make reliable contact with the top/bottom surfaces of the model.

rburema, 2019-08-09:
Hi @AbeFM , Firstly: This is basically what 'stair stepping' is used for: Taka a look at the 'stari step height' and 'stair step width' settings.

Otherwise, it isn't clear at all what a slanted wall is vs. a slanted bottom: The part you want support on might register for someone else as a wall they want to keep away from. (And adding an extra value for what is and isn't a wall isn't really that easy to add with how the algorithm works right now.)

AbeFM, 2019-08-09:
I don't feel you fully understood my issue. I believe you closed the issue without due consideration. I have been working on this for well over a year and dismissing it out of hand was perhaps a little unjustified? I have played with stair height a bit, and I did not find them to make any difference. If you have some recommended settings I would be more than happy to try them if it will resolve the issue.

I would like to know how to get support to stick when there is a large gap between it and the part.

[image] (stair step height 0, step width 0)

If I set XY distance to a very small number, I can get sticking or surface defects when extrusion from one nozzle encroaches on existing extrusions from the other material.
Setting XY distance large prevents adhesion (in the area marked in purple).
Due to the Min XY distance not being respected, there are gaps MANY layers tall where the bottom of the support should be. Where do all those strings end up? 

If you are still unconvinced, I can post pictures of what happens to printed objects as the XY value gets larger - I would hope that just by considering the image above it would be obvious that supports WILL NOT WORK if they are not attached to the model or some solid object.

If I'm missing something, please enlighten me. I have tried values for stair step height and width (from 0, 0.001, 0.01 0.1, 0.5, 1, 2,3...) and been unable to show an improvement in behavior - but I would be grateful to learn if I'm missing something.
[image]
Stair Step Height 1, Step Width 0

[image]
Stair Step Height and Step Width both 0.5
Again, I felt this was a reasonable issue, there is already work in what I consider to be a related area:
https://github.com/Ultimaker/Cura/pull/4807
From what I read there, it seems like the desire was to adjust the distance on both the floor and the roof

> New Settings:

"Support Interface Closest Layers Count"("Support Roof Closest Layers Count", "**Support Floor Closest Layers Count**") - number of the top interface layers closest to the part / bottom interface layers closest to the support for separately controlled line distance.
"Support Interface Closest Layers Line Distance"("Support Roof Closest Layers Line Distance", "**Support Floor Closest Layers Line Distance**") - value of the separate line distance.

Additionally there are other efforts by respected contributors in their own builds to solve the issue.

Ghostkeeper, 2019-08-13:
It seems though that the Minimum Support X/Y distance is held fine. The distance seems to be higher than 0.1mm as the setting says.

Should it be holding on to the Support Z Distance there? It seems that Support Distance Priority is the setting that is not being applied properly on the bottom; it seems to just subtract the model in both X/Y and in Z directions from the support.

smartavionics, 2019-08-13:
Hi @Ghostkeeper , I have been working on fixing this, see https://github.com/smartavionics/Cura/issues/12 for related discussion (if interested).

AbeFM, 2019-08-13:
I've been testing it, got my first prints with both the supports stuck and the sides not sticking in a while.

I'll keep posting at [smartavionics#12](https://github.com/smartavionics/Cura/issues/12) where this is in active development. At first blush it is fixing the issue, I don't know if it is causing side effects yet that require review.

closed 2019-12-20 by rburema