Ultimaker/Cura #6124: [4.2.1] Can't Import Stock Profiles.

eastabrooka, 2019-08-01:
Application version
4.2.1

Platform
Win10 x64

Printer
Ender 3 

Reproduction steps
1) Get https://www.3dprintedtabletop.com/download/cura-cr10-ender-3-terrain-profile-v-1-3/
2) Get Cura 4.2.1 
3) Open Cura, Say you have an Ender3 
4) Import the Ender3 profile. 
5) Error ! 

Actual results
[image]

Expected results
It should import the profile

Additional information
Pls Help !

JohnEdwa, 2019-08-02:
This is because of the Creawesome merge in Cura 4.2. The profile expects your Ender 3 definition to inherit the ``fdmprinter`` definition, but it now inherits a ``creality_base`` and *that* inherits the ``fdmprinter``.

Any Creality printer profile made before 4.2 will not be compatible and even if you could make it import by editing it, because the base definitions and profiles have vastly different settings, it most definitely wouldn't work properly anyway.

It would need to be recreated for Cura 4.2 from scratch.

eastabrooka, 2019-08-02:
Oh, Well, That probably means im going to be back on Cura 4.0 until the 3dprintedtabletop people make a new profile. 
Breaking backward compatability across all new adopters is probably a great way to make people lose their fave community profiles. 
Is there any way to force Cura 4.2 to recognise Ender3 Printer as type fdmprinter ?

JohnEdwa, 2019-08-02:
>Is there any way to force Cura 4.2 to recognise Ender3 Printer as type fdmprinter ?

No, and it wouldn't work anyway. The way Cura creates the profiles is stacking a bunch of overrides on top of a base file, and while fdmprinter is the one on the bottom and the 3dtt custom profile is the on on top, the files in the middle have changed.

There are some people (like me) who are poking at creating a legacy definition that would allow this, but it'll take some time to get it right.

zachshallbetter, 2019-08-11:
Is there a reader of some kind to help us pull the info out of the profiles. When the update occurred, it took the profile settings with it.

Ghostkeeper, 2019-08-12:
In the configuration folder (Help -> Show configuration folder) the profiles are under quality_changes and they are text files, so you can open them up with Notepad++ or whatever you prefer.

.curaprofile files are secretly .zip files so you could rename them and open them up with an archiver application like 7-zip. Inside you'll find the same type of text files, one for each extruder and one for global settings.

The problem is that Cura doesn't know which of your profiles are made for Creality printers and which are not. The profiles for Creality should get the "definition" field under "[metadata]" filled with "creality_base". The others should have " fdmprinter" there. You could also just copy all of the entries under "[values]" over to a different profile.

kazooless, 2020-09-29:
@Ghostkeeper As more and more printers are upgraded to use the variants, this "problem" is going to come up more. I just installed 4.7.1 and the TronXY printer will not use the previous profile because there are newly named profiles for it now. For anyone who doesn't understand the under-workings of Cura, this is confusing.

As I was thinking about this, I though of a possible solution. What about having a default "global" set of profiles which are available and accessible to all printers regardless of custom variants but they are hidden if they have their own set programmed?

Ghostkeeper, 2020-10-02:
> As I was thinking about this, I though of a possible solution. What about having a default "global" set of profiles which are available and accessible to all printers regardless of custom variants but they are hidden if they have their own set programmed?

That's basically what we have. There is a default "global" set of profiles here: https://github.com/Ultimaker/Cura/tree/master/resources/quality (scroll to the bottom). These profiles are accessible to all printers regardless of custom profiles unless the printer sets the `has_machine_quality` metadata entry to `true`.
Custom profiles are classified to belong to a printer if they are created while a printer with `has_machine_quality` set to `true` was active. Otherwise they are classified to belong to all printers that have it set to `false`. The problem here arises from when an update changes the `has_machine_quality` metadata. We don't know which of those custom profiles to classify as belonging to that printer then, since previously they were global.

kazooless, 2020-10-03:
> These profiles are accessible to all printers regardless of custom profiles unless the printer sets the has_machine_quality metadata entry to true.

And that is what is causing the problem. I'm still seeing it come up in different help forums and expect we'll see more people confused and afflicted as more and more printers are upgraded to use the new format. But since it is pretty much already built-in, as you say, then this might be easier than I thought. I'm guessing the reason the global profiles are not available when `has_machine_quality` metadata entry to `true` is the possible conflict if the same name is used in both the custom quality and the global quality. If that is the case, then wouldn't a simple logic check work? Something like:

If `quality_type` doesn't match any `custom_quality_type` but does match `global_quality_type` then use `global_quality_type`

or a little more complicated

if `setting_version` is "prior to version where `has_machine_quality=true`" then use `global_quality_type`

Ghostkeeper, 2020-10-06:
> I'm guessing the reason the global profiles are not available when `has_machine_quality` metadata entry to `true` is the possible conflict if the same name is used in both the custom quality and the global quality.

No, the reason is that not all global quality levels are supported by all configurations of all printers. There is a wide range of global quality profiles available globally, from 0.06mm to 0.6mm layer thickness. For instance using an Ender 3 with a 0.2mm nozzle, you'll only get two quality levels: 0.08mm layers and 0.12mm layers. Because printing 0.6mm layers with a 0.2mm nozzle is just not going to work. If printers have their own set of quality profiles, they can also specialise which quality levels are available for each combination of nozzle+material.

closed 2024-12-07 by github-actions[bot]