Ultimaker/Cura #6126: [4.2.1] Dis-allow overwriting profile on load

AbeFM, 2019-08-01:
**Is your feature request related to a problem? Please describe.**
I loaded someone's saved project, and it overwrote my entire print profile which seems to be lost forever now.

**Describe the solution you'd like**
ONLY allow projects to be loaded in with a new print profile. If someone wants to save over the profile that they happened to be using the last time they sliced something, they can do it with the Update command. 

**Describe alternatives you've considered**
Remembering what will happen if you just accept the defaults. When I try, sometimes I don't remember, and I lose a week of work.

**Affected users and/or printers**
Anyone who loads a saved file.

**Additional context**
[image]

I wouldn't have expected this to overwrite my current profile with a different name. It seems like the printer settings remain, but not my print profile.

nallath, 2019-08-02:
Do you just want the default to be switched? Because the create new does what you describe as the expected behavior.

AbeFM, 2019-08-02:
Yes, please change the default.

Even so: I think it is too easy to do too much damage that way. Either one more "Are you sure you want to overwrite profile [PROFILE_NAME]?" or else just load it in, if they want to overwrite their profile with something from an entirely different printer (I have a LOT of "custom FFF printer"s) with something they just downloaded online to help a forum poster out.... They can explicitly do so with the "Create Profile from existing settings" command, and picking it.

Perhaps another option would be "Update Existing" makes you select which existing profile you want to overwrite. That, at least, would make it intentional.

Hope this all helps. :-)

Ghostkeeper, 2019-08-09:
It only overrides if the profile name is exactly the same, so downloading something from online to help a forum poster out should not be destructive unless you happen to have the exact same names. I do this quite often, you see!

I don't think adding an additional pop-up would be a good solution. People don't read pop-ups (as is apparent from this ticket as well) and tend to hate having to click through them.

We can consider changing the default though. But the disadvantage of that is that loading a 3MF file then always creates a new profile. So if you drag the same project file onto your build plate 5 times it'll create the profiles `My Profile`, `My Profile #2`, `My Profile #3`, `My Profile #4` and `My Profile #5`. This makes switching between project files annoying.

AbeFM, 2019-08-09:
Huh. Thanks for the explanation. Apparently I misunderstood what was going on. I thought my (differently named) profile got overwritten, but it did not!

Totally a misunderstanding on my part. I agree, overwriting a profile with the SAME name helps the flow for the user. I see no reason to change anything, except send me a postcard saying RTFM. :-)

Thanks for clearing that up!

AbeFM, 2019-08-09:
On a related note: It would be nice to bring it into memory *but not save it at all*. Checking some random guy's post online clutters up my printers and profiles. Can it all be done 'in memory'?

I can re-open or start a new thread if it makes more sense.

closed 2019-08-09 by AbeFM

Ghostkeeper, 2019-08-11:
There's no feature for that right now. It's a bit complex since that would mean that Cura would need to keep track of 2 profiles with the same profile ID.

Cura does its very best to save all of your profile state always.

Of course you could try stuff like running Cura as a different user or something but then it'd just be simpler to simply tell Cura to create a new profile and delete it afterwards.

AbeFM, 2019-08-12:
Reasonable. Thanks for the discussion.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

Ghostkeeper, 2019-08-13:
> I see no reason to change anything, except send me a postcard saying RTFM. :-)

Maybe I'd do that if we have a proper manual... The contributor portal is broken at the moment though so I can't upload my update for the SettingsGuide :cry:

AbeFM, 2019-08-13:
Shoot, even a non programmer like me could help there.

With a bit of "hand holding" it wouldn't be impossible for me to submit
fixes instead of complaints.
But if you can't, I can't. :-)

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes