Ultimaker/Cura #6127: Re-org the Settings/profile dropdown

OldGaf, 2019-08-01:
I have a LOT of profiles and I have to scroll and scroll to the bottom of the list to manage them.

**Describe the solution you'd like**
Please move the "Create, Update, Discard, Manage" section on the Settings/Profile dropdown to the top. Putting menu selections at the bottom of a growing list is not the best idea.

**Describe alternatives you've considered**
There are no alternatives.

**Affected users and/or printers**
Everyone will benefit from this.

**Additional context**

smartavionics, 2019-08-01:
Something like this, perhaps?

[image]

I shall test this some more and if there's no problems will at least put it into my own releases. If the Cura devs like it to I will submit a PR.

OldGaf, 2019-08-01:
wow that is EXACTLY what I was looking for... makes MUCH more sense!

smartavionics, 2019-08-01:
I agree, I had thought of doing this some time back but never got around to it before.

You can give the materials menu the same treatment.

OldGaf, 2019-08-01:
Was this hard to do? I have done a fair amount of scripting etc but nothing for Cura.

smartavionics, 2019-08-01:
No, not hard, but I'm not familiar with the Cura UI and so couldn't make any big changes. This change just involved moving some script lines about.

````
diff --git a/resources/qml/Menus/MaterialMenu.qml b/resources/qml/Menus/MaterialMenu.qml
index ba35a160b..c2eea8216 100644
--- a/resources/qml/Menus/MaterialMenu.qml
+++ b/resources/qml/Menus/MaterialMenu.qml
@@ -37,6 +37,16 @@ Menu
         enabled: updateModels
     }
 
+    MenuItem
+    {
+        action: Cura.Actions.manageMaterials
+    }
+
+    MenuSeparator
+    {
+        visible: favoriteMaterialsModel.items.length > 0
+    }
+
     MenuItem
     {
         text: catalog.i18nc("@label:category menu label", "Favorites")
@@ -129,11 +139,4 @@ Menu
     ExclusiveGroup {
         id: group
     }
-
-    MenuSeparator {}
-
-    MenuItem
-    {
-        action: Cura.Actions.manageMaterials
-    }
 }
diff --git a/resources/qml/Menus/ProfileMenu.qml b/resources/qml/Menus/ProfileMenu.qml
index 68260f250..411f6c36c 100644
--- a/resources/qml/Menus/ProfileMenu.qml
+++ b/resources/qml/Menus/ProfileMenu.qml
@@ -11,6 +11,13 @@ Menu
 {
     id: menu
 
+    MenuItem { action: Cura.Actions.addProfile }
+    MenuItem { action: Cura.Actions.updateProfile }
+    MenuItem { action: Cura.Actions.resetProfile }
+    MenuSeparator { }
+    MenuItem { action: Cura.Actions.manageProfiles }
+    MenuSeparator { }
+
     Instantiator
     {
         model: Cura.QualityProfilesDropDownMenuModel
@@ -73,12 +80,4 @@ Menu
     }
 
     ExclusiveGroup { id: group; }
-
-    MenuSeparator { id: profileMenuSeparator }
-
-    MenuItem { action: Cura.Actions.addProfile }
-    MenuItem { action: Cura.Actions.updateProfile }
-    MenuItem { action: Cura.Actions.resetProfile }
-    MenuSeparator { }
-    MenuItem { action: Cura.Actions.manageProfiles }
 }

````

OldGaf, 2019-08-02:
[image]

OldGaf, 2019-08-02:
Perfect :)

diegopradogesto, 2019-09-10:
I don't know if it makes more sense. I will ask @Yian-Lai (our UXer) to take a look at it.

OldGaf, 2019-09-10:
It absolutely make more sense. :)