Ultimaker/Cura #6128: Does not render some gcode correctly

probonopd, 2019-08-01:
**Application version**
4.1.0 AppImage

**Platform**
Linux

**Printer**
Creality Ender 3

**Reproduction steps**
Load attached gcode and compare Cura's rendering to e.g., Repetier Host's rendering

**Actual results**
See right-hand side of screenshot

**Expected results**
See left-hand side of screenshot

**Additional information**

[image]

[wing.zip](https://github.com/Ultimaker/Cura/files/3458665/wing.zip)

rburema, 2019-08-08:
In the color-scheme legend, turn on 'travels'. It still won't look as you probably intented, but at least you can see all movements.

If the extrusion rate is low enough, it is seen as a travel-move.

Ghostkeeper, 2019-08-11:
The reason this is going wrong is because Cura is unable to detect the layer height when the layers are not planar. The layer height is necessary because the g-code only contains the amount of extrusion and line length. Cura only knows the relation of width * length * height = volume, so with 2 of these variables set, the other two are ambiguous. As such, if the layer height is guessed too high, the lines will be too thin. If the layer height is guessed too low, the lines will be too thick.

I think that Repetier may be assuming a different layer height as default, making it render better or more consistently to the eye. Otherwise they could be using a different sort of algorithm that is able to detect locally how high a layer (or just part of it) is, but I wouldn't know what that is since it almost requires physical modelling of how far the line will droop down to get there.

Going by my guess that Repetier is using a fixed layer height for all lines, I don't think that'd be something that we want to implement since it'd cause problems with our own Adaptive Layer Height g-code. If they're doing something more clever I'd be curious.

closed 2024-12-07 by github-actions[bot]