Ultimaker/Cura #6129: [4.2.1] Project don't retain objects position

GMagician, 2019-08-01:
**Application version**
4.2.1

**Platform**
Window 10

**Printer**
Custom but I think it's not the cause

**Reproduction steps**
Load attached project. Move objects and save. Reload project

**Actual results**
Objects don't preserve their position

**Expected results**
Objects layout unchanged

**Additional information**
STL was one for 3 object. These were be splitten by mesh tool

rburema, 2019-08-08:
Sorry, I can't reproduce this (with UM3). Just to confirm: you Save rather than Export right?

GMagician, 2019-08-08:
In project I move object then save project. When I reload project positions are wrong.

Vandrasc, 2019-08-08:
It also works for me! Unfortunately, the project file contains an unknown machine so I cannot say if it is related to that, could you please try with a different machine, just to confirm that the issue is not related with your own machine definition? The Custom FFF printer will do. 
Thank you!

GMagician, 2019-08-08:
Tested with ultimaker5...same issue. I also remove "mesh tools" that I used to split objects...the same
[2019-08-08 17-10-51.zip](https://github.com/Ultimaker/Cura/files/3482478/2019-08-08.17-10-51.zip)

Vandrasc, 2019-08-09:
In your video, I do not see the Save project window popping up when saving the file but I guess you left that out!? 

Really weird, I still cannot reproduce, this is my reproduce steps:

1. Add an Ultimaker S5 and load the handle.3mf project file
2. Move the parts around (make sure to be located inside of the build volume) 
3. Save the project File>Save.. --> the save project window pops up and I click [Save]
4. I reopen the last saved project file with File > Open recent...

GMagician, 2019-08-09:
In my video there is no popping window at all, they haven't been recorded by the program I used don't know why, but you may clearly see the "saved" and "loaded" window.

Only difference in what I do is step 4. I don't use recent but open and browse for file

Ghostkeeper, 2019-08-11:
I've seen this before and I can reproduce the issue in Cura 4.2.1. Here are the reproduce steps:

1. In the Marketplace, install the Mesh Tools plug-in and restart Cura.
2. Load a model that consists of multiple pieces, such as [this one](https://github.com/Ultimaker/Cura/files/3490042/separate_meshes.zip)
3. Go to Extensions -> Mesh Tools -> Split into parts.
4. Move one of the meshes.
5. Save the project file.
6. Clear the build plate.
7. Load the previously saved project file.

Something with that mesh tools plug-in sets an origin strangely or maybe the meshes are placed as child nodes of the same group node or something. I don't know and don't have the tools on this computer to debug that. But whatever it sets, Cura doesn't save its state correctly in the project files.

Ghostkeeper, 2019-08-12:
So this is causes by the Mesh Tools plug-in only, as far as we know. It's setting the origin of the mesh to something arbitrary. Cura should in theory be able to correct that though. @nallath investigated the issue back then but it was not simple to fix from Cura's side. So we're going to close this and you'll be at the mercy of that plug-in to get it fixed. Sorry.

Workaround is to save the meshes to STL and then re-load them.

closed 2019-08-12 by Ghostkeeper

rondlh, 2022-02-12:
I still see this behavior in Cura 4.13, some object seem to move by about 10mm after reloading the project.

mombof, 2023-03-02:
I had the same issue recently but it definitely happened after I used meshtools "split mesh into parts". Try using meshtools "reset origins to center of mesh", save close and reopen

kluke3, 2023-04-08:
This is still an issue in Cura 5.3.0. and it drove me crazy until I found this thread. The fix works that @mombof suggested still works. Select all the objects on your bed, then go to Meshtools, then click "Reset Origins to Center of Mesh".

kluke3, 2023-04-08:
After leaving the previous comment, I noticed there was an update for the Mesh Tools plugin. I thought the issue might have been fixed. It appears that the issue is still not fixed in version 3.7.1. To test it, I used a CAD program, to created a single body consisting of 4 cubes. I exported it as a single STL file. Then I went into Cura, and split it, then moved the cubes around. I saved the project and closed Cura. When I opened the project in Cura, the cubes were not in their original positions, and two of them were actually under the bed to where it could not be seen from the top.

lhofinger, 2023-05-28:
I've just the same problem (version 5.3.1) without any plugins. When I reload a project with two models, objects are not at the same position.
[image]
[image]