Ultimaker/Cura #6130: The print settings does not take over the build plate temperature from the material

albbino, 2019-08-01:
I create a new material with default printing temperature of 240°C and a default build plate temperature with 85°C

If I select this material and go to the print settings of the project the printing temperature is 200°C.
There is a small button to click with an fx on it. If I press the button the value is changing to 240°C

The build plate temperature does not have this button. Why? How to take over the settings in the material of the build plate temperature.

rburema, 2019-08-08:
I can't reproduce this in 4.2.1 can you save (not export) your project as a 3mf and zip it here? Maybe we can have a look at what's going on.

albbino, 2019-08-09:
[example.zip](https://github.com/Ultimaker/Cura/files/3486927/example.zip)

Example and screenshot added.

Ghostkeeper, 2019-08-11:
This is because the [normal quality profile](https://github.com/Ultimaker/Cura/blob/master/resources/quality/anycubic_i3_mega/anycubic_i3_mega_normal.inst.cfg) sets the build plate temperature to 60 degrees and the printing temperature to 200 degrees. The quality profile overrides the material profile, so if the quality profile changes the temperature settings, your customised material will no longer have an effect on those settings.

It's a bit weird of the Anycubic profiles to set the temperatures to be fixed like this. They're basically saying that you must always use 200/60 degrees regardless of what material you're printing with. It would've been better to just say that material profiles are unavailable on this printer, like with the Ultimaker 2. Those setting values are the same in every quality profile anyway!

I don't know why the fx button doesn't show for the bed temperature. As far as I can see, it should. We have an open ticket to rework the fx button to allow resetting values from user-created profiles instead of its current behaviour. But it'll take a while before we get to that since it doesn't get a lot of priority.

Ghostkeeper, 2019-08-13:
We're closing this because it's expected behaviour.

We can't make changes to the profile without testing them and we can't test changes to the profile without the printer itself. So since we don't have every printer on the market, we need to rely on other people to make those changes. If you think the profile is wrong, please make a correction!

closed 2019-08-13 by Ghostkeeper

MattiasEppler, 2019-08-14:
> If you think the profile is wrong, please make a correction!

Where to find the profile?

Ghostkeeper, 2019-08-14:
There are 3 profiles that have these 200/60 temperatures in them. They are all three in this folder: https://github.com/Ultimaker/Cura/tree/master/resources/quality/anycubic_i3_mega

albbino, 2020-02-06:
Hi. Have an additional question: You wrote

> We have an open ticket to rework the fx button

Where can I find this ticket?

fieldOfView, 2020-02-06:
I'm guessing you can't; it will be an internal ticket.