Ultimaker/Cura #6131: Pause at Height Cura 4.2

sanpesa, 2019-08-01:
**Application version**

**Platform**

**Printer**

**Reproduction steps**

**Actual results**

**Expected results**

**Additional information**

Hi, I wanted to report a bug that I discovered today in Cura 4.2 when you use the paude at height option on my ender 3 printer.When you use this option to make a filament change and when restarting, the nozzle is very separated from the last printed layer and the filament is not stuck as it should. I have tried to print it twice by configuring the file both times and both times it has made the same error.I wanted to know if there is any solution or if it is the first news you have, that you take note of it.

P.D .: I am using the Cura in Windows 10 Home x64

A greeting.

NickGlennon, 2019-08-02:
I am also having this same problem Cura v4.2.1 Windows 10 Home x64 on a CR-10 mini.  I have two layer changes 5 layers apart happening (0.8mm), and the first one behaves just like this.  I also resliced the file, same behavior.  Not sure if its trying to continue off the next layer change or it's an arbitrary value that's being added to the Z height.

Kenpowarrior, 2019-08-03:
Update 4.2 makes pause at height on ender 3 restarts to high instead of starting on correct layer. Maybe downgrade cura to 4.0 to get it working right again?

sanpesa, 2019-08-04:
That option worked well for me in version 4.1, and I have already reinstalled it

Kenpowarrior, 2019-08-04:
I reinstalled 4.0 and it all works fine again.

pseudex, 2019-08-07:
I'm not sure how it was on Cura 4.1 but I think they moved the Z move to the previous layer:

```
;MESH:NONMESH
G0 F300 X117.34 Y121.017 Z0.44
G0 F6000 X102.796 Y132.552
;TIME_ELAPSED:188.682909
;LAYER:1
M140 S60
M106 S85
;TYPE:WALL-INNER
;MESH:PawPatrol Ryder MultiColor v5_wei??.stl
G1 F1500 X102.453 Y132.144 E147.98385
```

the Pause script relies on the first Z move in the next layer.

sanpesa, 2019-08-07:
Then I understand that it is something that the developers of the program have to fix, and we cannot do anything to solve it, right?

Kenpowarrior, 2019-08-07:
Only thing you can do is downgrade to 4.0 or 4.1 where you know it still works.

ramonmelo, 2019-08-09:
Same problem here. I've been using Cura 3.6 for a while and decided to upgrade to 4.2, and the first piece I've made was using the "Pause At Height", which failed 3 times.

I hope the devs could fix this asap.

Ghostkeeper, 2019-08-11:
Duplicate of https://github.com/Ultimaker/Cura/issues/5924.

closed 2019-08-11 by Ghostkeeper

ItsDLite, 2019-08-15:
Your looping the entry and have closed both! Please fix this you do need to downgrade to 4.0 to resolve the issue

Ghostkeeper, 2019-08-16:
No, #5924 is still open! We do intend to fix this.

hussainsail2002, 2019-08-19:
Hi ,

I am having the exact same problem, My print starts exactly 1 mm above the required position

@Ghostkeeper I think you closed #5924 referencing this thread and you closed this one referencing that ? I am not sure, just checking.

Thanks 
Hussain

sanpesa, 2019-08-20:
@Ghostkeeper in version 4.0 the error is already solved because I made pause at hight a couple of times since I opened this thread, and it works perfectly.  I mean that the thread that should be closed is the other one and not this one, apart from the fact that they are threads for different versions and I do not consider this duplicate.

Ghostkeeper, 2019-08-20:
> @Ghostkeeper I think you closed #5924 referencing this thread and you closed this one referencing that ? I am not sure, just checking.

No, #5924 is still open. Nobody ever closed it.