Ultimaker/Cura #6132: Dark Shaded Area Needs To Have A Manual Override, Especially For Multiple Extruders

Scottlegend, 2019-08-01:
The Auto Dark Shaded Area To Reduce Print Area Size Along The Perimeter For Safety Or Printer Abilities Needs To Have A Manual Override Setting For All Printers To Adjust What The Printer Can And Can't Do. Especially For Dual Or Multiple Extruders As There Are Many Circumstances Where All Extruders Can Reach The Entire Build Area Yet Cura Will Not Allow This. In My case I have 2 Extruders Where 1 Is A Direct Extruder And The Other Is A Bowden Set Up And Every Time I Need To Change The Nozzle Size I Need To Re-calibrate The Offsets As They Are Independent Which I Use Cura To Do So As It Is Much Faster Than Adjusting The Firmware On The Printer. I Am Not The Only One With This Issue, I have Done Some Research And There Isn't Anything To Address This. I Would Add A Manual Button That Allows All Greyed Out Areas To Be Turned Off Or Manually Adjustable. 
Kind Regards.

rburema, 2019-08-08:
So let me see if I understand this: Going into 'Machine Settings' each time is a hassle for you, since you have to do it a lot, and you'd like a button to disable the detected unavialable areas on the buildplate.

Personally I think this creates a lot more hassle than it's worth; people will have it turned off, forget about it, then complain that the nozzle collides with the model, or that support/adhesion is printed outside the buildplate or that the model intersects the prime-tower/prime-blob, etc., etc.

Ghostkeeper, 2019-08-11:
There is something to address this.

In the printer definition there is a metadata entry `nozzle_offsetting_for_disallowed_areas` that allows you to configure whether the shaded areas around the sides move along with the nozzle offset (i.e. the volume is bound by what the nozzles can reach) or not (i.e. the volume is bound by what the print head can reach). As an example, take the Ultimaker S5 definition:

https://github.com/Ultimaker/Cura/blob/d703e48007dc201ffda108dbe46077eb8ae949e4/resources/definitions/ultimaker_s5.def.json#L23

This is not configurable in the UI since it is not common that you need to change this unless you built your own printer. If you're using a printer with grey areas at the sides of the volume due to the nozzle offset, but those areas are reachable by both nozzles anyway, then the author of that printer definition has made a mistake. If we know which printer you're using, we can maybe make a correction to your definition?

If you did build your own printer that has this property, then I think you'll be fine with putting the metadata entry there yourself.

closed 2019-08-11 by Ghostkeeper

Scottlegend, 2019-08-12:
Cheers for the response, I have found a way around it by adding the nozzle offset through G code start up script and just made the build plate bigger for what the offset is. Although I have 2 hotends they are both mounted on the same carriage. they are both mounted at the same nozzle height. I get the Z axis clearance for hitting parts however it is just X and Y that i wish was customizable for the shaded area with an instant where both nozzles can pass over the entire build area i would like to turn that off as that is my set up and as an alternative add a positive limit to stop it crashing, obviously i would manualy check first it can reach there to prevent damage. There cannot be a crash with the print by changing x and y shaded area for the outside of the bed, leave the parts shaded just not the bed size is what i am getting at.

Im not sure what you mean by "the author of that printer definition has made a mistake"

I get that not many users need this however in my case, i use the entire build volume edge to edge. If you did set this up as an option i don't see how it will cause problems if done correctly.

I started with a Cr10 although the only thing left of it is the frame.

Kind regards
Scott

From: Ghostkeeper
Sent: Monday, 12 August, 12:40 am
Subject: Re: [Ultimaker/Cura] Dark Shaded Area Needs To Have A Manual Override, Especially For Multiple Extruders (#6132)
To: Ultimaker/Cura
Cc: Scottlegend, Author

Closed #6132<https://github.com/Ultimaker/Cura/issues/6132>.

Ghostkeeper, 2019-08-12:
> Im not sure what you mean by "the author of that printer definition has made a mistake"

The author of the CR10 profile didn't set that `nozzle_offsetting_for_disallowed_areas` to `false` so if the CR10 has its print volume limited by the build plate size rather than the print head's movement, that would be wrong of the author of that profile.

However it seems like you practically modded your printer to something completely different, so the author of that printer profile is not at fault here, really. We've always assumed that if you built your own 3D printer, adjusting a JSON file shouldn't be that hard comparatively.