Ultimaker/Cura #6134: No Support Structure

Splodgeink, 2019-08-02:
**Application version**
4.2.1
**Platform**
Windows 10 64bit - latest updates installed

**Printer**
Ender 3 Pro
**Reproduction steps**
See pictures
**Actual results**
See Pictures
**Expected results**
Using preview in 4.1 I could see the support structures. These weren't there on 4.2.1
**Additional information**
See above

Splodgeink, 2019-08-02:
[image]
[image]

Ha! New to programming and what I had typed in is not there! (still in those funny comment characters!)

Cura 4.2.1 is not generating support structures despite the setting being selected. It is working fine with 4.1 as I can see them in preview, but not in 4.2.1. I was printing https://www.thingiverse.com/thing:3207754 with the recommended settings.

Splodgeink, 2019-08-02:
Found the edit function...

JohnEdwa, 2019-08-02:
Even if you have supports on, the default setting for the Creawesome profiles is to have them limited to "Build Plate Only", which is a stupid idea.

2-k, 2019-08-02:
This is a major issue, I have a part that clearly needs support it's not going to get it with Cura 4.2, super annoying as this worked yesterday. The issue is clearly with the conical support function. Turned off support shows up, guess that's why it's still in the experimental section.

Splodgeink, 2019-08-02:
Perfect print with 4.1! No changes to any profiles.

Splodgeink, 2019-08-02:
[image]

Ghostkeeper, 2019-08-11:
In Cura 4.2, if you set Support Placement to Everywhere, does that:

1. Fix the problem, and
2. allow support to be removed from the model with reasonable ease?

I think setting the default to Everywhere is probably better from what we've heard people say. But we might need to adjust the Support Bottom Distance. So I have another question: What value for Support Bottom Distance gives the best results?

2-k, 2019-08-11:
The funky/no supports are caused by the conical support function. If you have it on then they 99% of the time won't show up. 
I got them to show up once when I set support to everywhere and 0° but never would slice supports in again. This is probably why it's still in the experimental section of the settings.

Sorry I didn't seem to post the solution for this. I figured it out shortly after posting this and must have put the answer on a different one I guess.

2-k, 2019-08-11:
All values should be based off your nozzle diameter and layer hight. For supports then you have to factor in removability, which is different for every printer, print and filaments. My philosophy is under extrude supports and over extrude infill.

closed 2020-06-19 by no-response[bot]