Ultimaker/Cura #6136: Where is the prime tower and Z Speed

parker1ray, 2019-08-02:
**Steps to Reproduce**

**Actual Results**

**Expected results**

**Additional Information**

Vandrasc, 2019-08-05:
Hi @parker1ray , please use the bug template so we can try and reproduce the issue in order to help you. 

Thank you!

parker1ray, 2019-08-05:
I just rolled Cura back to version  4.1 It seems to work best with the
Geeetech A10M, and the tower has come back and is square. In 4.2.1 there
was no option to allow for a square tower. and the maximum Z speed option
is gone from settings.
The Geeetech printers pause and blob if you are not able to set the Maximum
Z speed to anything other than 0. The prime tower also showed back up after
I downgraded to 4.1. You must have had some issues because you rolled out
version 4.2.0 and within a week 4.2.1 was out.

Vandrasc, 2019-08-05:
Hi @parker1ray ,

Indeed, the maximum Z speed has been removed from the engine and frontend as there were some issue where even if the settings = 0 by default was causing an M203 command that would set the movement speed very high and It would also crash the printer firmware. It was decided that the setting was superfluous and thus removed but in the Printer Settings plugin, you can still find the maximum Z speed if you wish to adjust it. 

The Square prime tower has been removed and replaced by the round one, as we observed + received feedbacks that the adhesion and other issues related to the prime tower are fixed in the round one. Why do you prefer the square prime tower instead of the round one? What are the advantages when compared with the default round one? 

Yes, there was an issue where the slicing was blocked for some of the Creality printers if Support would be generated, this being the reason for 4.2.1. Also, we are a flexible team and it is not that odd of having multiple batches for important quick fixes.

closed 2019-08-05 by Vandrasc

parker1ray, 2019-08-05:
Thanks for the info. I do not prefer a round prime tower, but found that in
version 4.1, there was a huge issue with pausing and blobbing on an A10M
when maximum z-speed was set to 0. I messed with Cura settings for two
months and tried everything to get rid of the pausing and blobbing issues.
I finally found a very obscure post where the person instructed people to
set the Maximum Z-Speed to 35mm/sc. This fixed the issue and the printer
that had previously never printed one successful print began to print
beautifully. I upgraded to 4.2.1 and found that  with the round tower for
some reason I had sever under extrusion and the pause was back on the
circular tower. That is the reason that I have reverted back to 4.1. It is
working again as you can see from some prints that I did over the weekend.
I also reasoned that a square tower has more surface area than a round
tower which helped me with adhesion to the bed. I had several round towers
fall where the square ones did not. I will try the a newer version in the
future once enough new improvements have come along.