Ultimaker/Cura #6137: [4.2/4.2.1] I can't click on materials

Liger0, 2019-08-02:
**Application version**
4.2 and 4.2.1

**Platform**
Win10 x64

**Printer**
Creality cr-10s

**Reproduction steps**
I try to open the material tab

**Actual results**
I can't select any material, not from the favourites nor from its real dropdown. Trying to select a favourite material will select its dropdown menu. Trying to select a material from its dropdown won't do anything.

**Expected results**
Clicking on a meterial should let me able to edit it.

**Additional information**
https://youtu.be/uDgGOQsFy8g?t=31
Watch it from 0:31

REBU47, 2019-08-02:
Application version 4.2 and 4.2.1  Platform Win10 x64  Printer Anycubic Mega S  I have the same problem. I can not choose any material. Everything was OK in CURA 4.1

fieldOfView, 2019-08-02:
Do either of you have the Material Settings plugin installed? If so, please try if disabling that plugin and restarting Cura fixes the issue.

Liger0, 2019-08-02:
Unfortunately, in Cura 4.x there is no way to uninstall installed plugins. Even tho they result installed, the work "install" still appears instead of the word "uninstall".

fieldOfView, 2019-08-03:
In the Marketplace dialog, there’s an “Installed” tab (top right). There you can disable (uncheck) but also uninstall plugins. 

So you are saying you do have that plugin installed?

Liger0, 2019-08-03:
Yes, I have it installed. But as I said, since 4.x there is no button to uninstall plugins

Vandrasc, 2019-08-05:
You can uncheck the box but you always need to restart Cura in order to make the changes in the packages. But yes, I can confirm that when the Material Setting plugin is installed the materials cannot be selected anymore and when removing the plugin the materials are selectable and editable again. 

@fieldOfView , do we need to align on this? Does the plugin need an update? Thank you:)

fieldOfView, 2019-08-05:
The plugin needs an update. https://github.com/Ultimaker/Cura/commit/914d2e98e977131b67a85131ba29aadf63fb4f91#diff-7350e9d08a06d9246ab8a568e3becc2b broke the plugin.

Liger0, 2019-08-05:
I do not see any checkbox. :(

Vandrasc, 2019-08-05:
Preferably to Uninstall the plugin, but if you want to give it a quick try...Marketplace > Installed and on the left side of the plugin description, you have the checkbox. 

[image]

nallath, 2019-08-05:
> The plugin needs an update. [914d2e9#diff-7350e9d08a06d9246ab8a568e3becc2b](https://github.com/Ultimaker/Cura/commit/914d2e98e977131b67a85131ba29aadf63fb4f91#diff-7350e9d08a06d9246ab8a568e3becc2b) broke the plugin.

Hmm, but that fix ensured that there was no longer a difference between the first start of the preference window and clicking the material slot (eg; In the first start it got a different item as the currentItem you got when clicking the slot).

fieldOfView, 2019-08-05:
@nallath, the plugin contains a modified MaterialsPage.qml, (*) based on the one from 4.2. I did not notice the file had been modified in 4.2.1, so the copy contained in the plugin does not have the setExpandedActiveMaterial function.

(*): Replacing the whole MaterialsPage with a patched one is the only way the plugin can replace the Print Settings widget with less naive ones that properly display floats, include validation, etc.

fieldOfView, 2019-08-05:
See also https://github.com/fieldOfView/Cura-MaterialSettingsPlugin/issues/3

nallath, 2019-08-06:
Ah, right. That makes sense.

Ghostkeeper, 2019-08-08:
Closing this as duplicate of https://github.com/fieldOfView/Cura-MaterialSettingsPlugin/issues/3 then.

closed 2019-08-08 by Ghostkeeper