Ultimaker/Cura #6138: Cura closes after start up :(

technologia-13, 2019-08-02:
A few days ago I downloaded Cura 4.2, everything warked excellent. Today I tried to open Cura and it opened in taskbar for just 1 sec and immediately closed, I have also noticed that after closing the program is visible in the task manager. I have tasted that problem on a few versions and all look the same. 
Cant find the solution

technologia-13, 2019-08-02:
So I've tried everything, nothing worked, then I turned off Avast...

closed 2019-08-02 by technologia-13

fieldOfView, 2019-08-03:
Do you have multiple monitors? If so, could you try renaming Cura.cfg from the configuration folder?

technologia-13, 2019-08-03:
No I don't multi monitors but when I wrote about Avast I meant that it worked after turning off Avast, still I don't know why that is the thing but when it is turned down Cura works when on it doesn't.

spikeeee, 2019-08-03:
Having the same issue with 4.2.1.  Runs once but then fails to open if closed.  

Found that Malwarebytes is blocking the program from running when Malware Protection was enabled.

Added Exclusions for the folder  
      c:\Program Files\Ultimaker Cura 4.2\
And Program Exclusions for:
       c:\Program Files\Ultimaker Cura 4.2\Cura.exe
       c:\Program Files\Ultimaker Cura 4.2\CuraEngine.exe

Now the program loads fine

technologia-13, 2019-08-03:
Exclusions didn't work for me, few minutes ago I even had to turn off Windows Defender.

Ghostkeeper, 2019-08-11:
Cura uses a local socket to communicate to CuraEngine. This is very common in Linux environments but not so common on Windows. Some virus scanners block it because it gets used sometimes by viruses. Viruses in general are not so common on Linux.

Could you attach a log file here? Our bug reporting template asks for it after all...

technologia-13, 2019-08-12:
Looks like log file is not updating when that happens and I have to add that its getting weirder all the time, few days ago I had to uninstall 7zip for cura to perform and now is seems to just sometimes work, next day not at all and few minutes later work again, without doing absolutely anything. ( Maybe uninstalling 7zip was unnecessary :/ )
Btw Ive noticed that old (I mean 15.something and below) versions works fine.