Ultimaker/Cura #6140: [4.2.1] Extreme lag when rotating a complex sliced model

Liger0, 2019-08-03:
**Application version**
4.2.1

**Platform**
Win10 x64

**Printer**
Custom dual extrusion

**Reproduction steps**
Slice a complex model

**Actual results**
In preview mode, rotating the model is almost impossible, it lags in an absurd way.

**Expected results**
It should rotate smoothly like in 3.x

**Additional information**

rburema, 2019-08-08:
We did offload some calculations to the shaders, as in most cases. it made the interface a bit quicker. I wonder if that could be the cause.

Could you maybe attach the model, so we're sure we're talking about/investigating the same thing?

Liger0, 2019-08-08:
[CCR10S_blinkfox stutter preview.zip](https://github.com/Ultimaker/Cura/files/3481831/CCR10S_blinkfox.stutter.preview.zip)

Ghostkeeper, 2019-08-11:
I have a very modern GPU, so it's running fine at 60fps while rotating. It should be mentioned that Cura is using more than 4GB of memory on my GPU while this model is in layer view. If your GPU doesn't have 4GB of memory left, it could be swapping memory in/out with your RAM and this is very slow.

I think it's somewhat expected to have some stutter on older GPUs if you are trying to display more than a gigabyte of g-code. I'm not noticing any difference in performance when using 3.6, but it'd be hard to notice since the frame rate is limited by the update frequency of my mouse cursor which is linked to the screen refresh rate.

Maybe the difference that you're seeing is because of different settings? For instance, maybe you didn't have Gyroid infill enabled back then? Gyroid is well known to create gigantic g-code files because of all the curves.

Liger0, 2019-08-11:
Aren't 8 unused GB of ram enough for it to be smooth?

Ghostkeeper, 2019-08-12:
Well that depends. For me it was. But it's still drawing some ~100 million triangles so VRAM is not the only factor then.

closed 2020-06-19 by no-response[bot]

Ghostkeeper, 2020-06-30:
In Cura 4.5 we've made some improvements to prevent the interface from re-rendering when you rotate the camera. That should improve the performance here.

jellespijker, 2020-06-30:
@Ghostkeeper Do we have all polygons in memory or just the outside ones? Such as shell and top layer and in the case of exposed infill (when we drag the layer-slider) an n-amount of visible "top" layers

rburema, 2020-06-30:
@jellespijker  Currently we only have lines, which then get sent to the shader-program. The shader uses a geometry-shader to create polygons out of the lines. This (the way it's done r.n.) is quite expensive (because it gets done each rendering step), and it's on my wish-list to investigate if it helps to move the polygon-creation to Python (bigger up front cost, greatly reduced rendering effort). The only problem with that is if the up-front cost would get so big as to annoy the user.