Ultimaker/Cura #6142: 4.2.1 - Purge line printing too far into my print area

Anarasha, 2019-08-03:
**Application version**
Cura 4.2.1 

**Platform**
Windows 10 

**Printer**
Creality Ender 3
[buggypurge.zip](https://github.com/Ultimaker/Cura/files/3464336/buggypurge.zip)

**Reproduction steps**
I encounted this bug by printing on my printer as normal, but with a fresh installation of Cura. I thought my print area had moved drastically to the right, but the objects were pretty much where they were intended to be. 
Upon inspecting my start GCode, I notice the following: 

> G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
> G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
And so on. 
This is a newly created Ender 3 profile with a vastly changed start GCode that starts the purge line 10mm into the printable area. 

**Actual results**
The purge line prints quite some into my build area 

**Expected results**
The purge line is supposed to print all the way to the left 

**Additional information**
This is a completely new installation, my SSD conveniently broke right before I was gonna upgrade, so my system is a completely fresh installation of Windows 10 as well as Cura.

saidzareh, 2019-08-04:
i  have the same problem and dont know how to fix

Anarasha, 2019-08-04:
> 
> 
> i have the same problem and dont know how to fix

It's simple enough to fix, it's because of the forced implementation of the over-hyped cre"awesome" mod that overwrites our carefully tinkered profiles and start GCodes. 
You go into your Cura startup code and change the violently and forcefully modified coordinates back to what they were before the forceful implementation. 
So basically, go to start GCode and find the part that starts with: 

> G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed 
And ends with: 
; Draw the second line 
And change that entire section to this: 
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line

Hopefully, Cura will be fixed so others won't have to fix all the changes back

nallath, 2019-08-07:
You could also make a non-forcefull pull request to fix it. That way there is no waiting on a hopefull fix because it will be there.

Anarasha, 2019-08-07:
Fixing it is easy, just change the coordinates in the start GCode. This is to let the Cura developers know that it's there as a default now. It appears to be a part of the now forcefully implemented Cre"awesome" mod. Although racking my brain I cannot think of a single reason why you want your purge line to print in the middle of where your print might well be. Haven't you always wanted a purge line ruining the bottom of your print?

Ghostkeeper, 2019-08-11:
Cura is putting the start g-code there properly, so this is not a bug and expected behaviour. So I'll close this ticket. You seem to have found a workaround that works for you.

If you have an improvement to the default profiles, please post your carefully-tinkered start g-codes here in a pull request and we'll help the Creality community by distributing them to their users :smiley:

closed 2019-08-11 by Ghostkeeper

Anarasha, 2019-08-11:
> Cura is putting the start g-code there properly, so this is not a bug and expected behaviour. So I'll close this ticket. You seem to have found a workaround that works for you.
>
> If you have an [...]

Okay, I accept that this is intended behaviour. Would you care to explain the reasoning to me though? I simply cannot, no matter how I twist my mind around it, imagine what the reason for a purge line in the middle of my print is good for. Why is it putting it there on purpose?

Anarasha, 2019-08-11:
@Ghostkeeper 

Okay, I accept that this is intended behaviour. Would you care to explain the reasoning to me though? I simply cannot, no matter how I twist my mind around it, imagine what the reason for a purge line in the middle of my print is good for. Why is it putting it there on purpose?

Ghostkeeper, 2019-08-12:
I don't know. I didn't even know it did that. Since I don't have every printer in my house I can only hear from others that it's doing so and only depend on others to maintain the profiles.

Anarasha, 2019-08-12:
@Ghostkeeper 
I get that. Please don't feel like I was attacking you, I absolutely did not intend that.
I was just hoping there was an explanation since this has been flagged as working as intended

Ghostkeeper, 2019-08-14:
I didn't mean to sound that sharp or attacked! Sorry.

But I don't know why it's drawing two lines there, either. I just know it says that explicitly in the start g-code, so the person that made that start g-code must've done that on purpose. Perhaps one line was too short to purge properly, and there wasn't any space to make it longer?

Anarasha, 2019-08-14:
@Ghostkeeper 
The two lines are entirely correct, it's been doing that for a while. And indeed, it's because one line isn't enough. 
I think I didn't phrase myself clearly - the issue I have isn't the two lines themselves. It's that ever since they were added they've printed 0.1mm into the printable area to be discreetly out to the side and that has worked really well. But in 4.2.1, this changed so they now print 10mm into the printable area - this means they will actually quite easily conflict with a print unless you mentally keep in mind that the 10mm on the left side are off limits for printing

Ghostkeeper, 2019-08-14:
The printer definition can define "disallowed areas", to help the user where they can't put their prints. It's what the Ultimaker printers use to define where the build plate clips are that hold the bed down. That's an option to deal with it.

Maybe the author of the definition just thought it wasn't a problem to have one line protruding from the bottom there. You can easily tear it off afterwards. Perhaps it was unreliable to have it be printed so far off the side.

Anarasha, 2019-08-14:
@Ghostkeeper I know what you're getting at, but I'm pretty sure this is unintentional. In version 4.1 and all the way back, it has always been to the side, and then in 4.2.1 it was changed to the new, conflicting way. 
Regardless, I thank you for your time - I'm gonna post it as a suggestion to be improved in the correct forum.

oofiksoo, 2019-09-07:
Wow, this is yet ANOTHER post of a cura guy saying "nope we wont change it" regarding an absolute issue introduced by the creawsome mod. I don't think you guys even read our posts here to you. The issue here is that Cura is placing the line at 10mm, instead of the default 2 mm that it should be starting at. So, Yes goatskeeper, it is placing the Gcode, but it doesn't mean the Gcode is correct. Cura always placed the line at the left edge, and now it places it 10mm from the edge. There is your bug. It is a simple variable change yes. But this GitHub thread, added to the 3 others I am a part of, ALL contain this exact example. And all you guys keep saying is "We're NOT GONNA FIX IT" "NOT MY PROBLEM" "I DONT HAVE A CREALITY PRINTER". I agree with the others whom have accused this of a conspiracy. It does seem as though you have not concern for the user experience of competing printer manufacturers.

nallath, 2019-09-09:
@Oofiksoo, You're right.  This is a live image of the Cura devs right now; 
[image]

All jokes aside, yes, we don't put priority on the print experience of competitors. To put it very bluntly, why would we? But just because we don't put priority on it, doesn't mean that we go out of our way to sabotage it. As i've said in every thread where you claim the sabotage conspiracy: Tell creality to start contributing, so it's no longer up to people doing this in their own time. They are also a company and they are failing in providing a good product (because 3D printing is more than just the hardware).

Anarasha, 2019-09-09:
@nallath For the record, I'm totally in your corner on that. The fact that you even open up your slicer and put in work to keep it viable for competing hardware is admirable - you have my eternal respect for that. I certainly don't consider you to be conspiring saboteurs like that other person, the idea is absurd. 

That said, the issue at hand here is clearly an unintended mistake and not working as intended. I would ask for one more glance at it - clearly it's not supposed to print a purge line 10mm into the print area, thus has to be a typing mistake when the Creawesome mod author input the purge coordinates - it should have said 0.1 and 0.2 like all prior versions but ended up saying 10.1 and 10.2. I can only imagine this is a mistake, a acatrer-brained moment of a content creator hard at work

nallath, 2019-09-09:
I'm more than willing to change it (because you're right, it's probably an oversight/whoopsie). But since we do want to make sure that it works, I would like to hear from multiple people that changing it to 0.1 mm actually works (Otherwise it might break stuff for a bunch of people again, so I'm just being cautious here).

Anarasha, 2019-09-09:
@nallath I mean it has been like that in every version of Cura before this one, is that enough? 
Otherwise I'll be glad to start a thread on the Ender 3 board to get some people to confirm it :-)

nallath, 2019-09-09:
I'd prefer it if a few more people test it, especially since a lot more things have also been changed. No matter how small the change, it's just safer to actually test it. The whole "This is a small change and it should work" are famous last words of quite a few engineers ;)

Anarasha, 2019-09-09:
Of course, I'll make a thread in a bit and post the link when at least 2 people have confirmed :D

Reality4DEvolution, 2020-07-07:
i have a creality cr10S pro v2.  just got it past month.  when it starts printing, the nozzle print starts on the left edge, goes vrom ZERO, up to middle of bed, then in a sideways "V", to center, and back.  i don't know if this is part of what you are describing, but indeed , this goes into middle of print area.  If that is done by CURA , then it may be related to this thread, and i believe would warrant modifying, so it is nt in common print area
OR, else, in cura, have an option t SWITCH the area where this gets printed.  REASON is that the area this prints in is gettins super-smoothh from scraping, compared to other aras of print bed;  might be good to rotate it around print bed  (have an option;  quadrant 1, 2, 3, 4)
anyway, just my two cents, thanks
https://www.dropbox.com/s/x32v0ltiy707nwe/CURA-nozzle-purge.jpg?dl=0