Ultimaker/Cura #6145: All support types are surrounded by a wall - V4.2.1

Eyos, 2019-08-04:
All support types now have a wall around them.
[image]
.

fieldOfView, 2019-08-04:
See the "Support Wall Line Count" setting.

nallath, 2019-08-07:
With support line count at 1:
[image]
With support line count at 0:
[image]

closed 2019-08-07 by nallath

Eyos, 2019-08-07:
Thanks for the information.I never used this option, it was always set to 0. When opening files in 4.2, this option was set to 1 automatically, without notice.Regards,Yossi