Ultimaker/Cura #6146: [4.2.1] Gap filling doesn't follow the holes and ends out of the model

Liger0, 2019-08-04:
**Application version**
4.2.1

**Platform**
Win10 x64

**Printer**
Custom dual extrusion

**Reproduction steps**
Use gap fill 

**Actual results**
it fills gaps out of the model in the air

**Expected results**
Fill gaps only in the model

**Additional information**
[gap filling in the air.zip](https://github.com/Ultimaker/Cura/files/3464921/gap.filling.in.the.air.zip)
[image]

Ghostkeeper, 2019-08-11:
This is already implemented. Try disabling the Print Thin Walls setting.

The bug that these lines are not properly joined into a long wall is already reported here: https://github.com/Ultimaker/Cura/issues/5445

closed 2019-08-11 by Ghostkeeper

Liger0, 2019-08-11:
It doesn't seem the same to me. In my case they don't fill the correct gap but are in a different position out of the model.

Ghostkeeper, 2019-08-14:
It's trying to fill something outside of the model?

Liger0, 2019-08-14:
Well yes as you see, it doesn't fill following the model but it goes far outside of the model mesh.

Ghostkeeper, 2019-08-15:
I'm finding it a bit hard to see. For one, I'm getting a different result with that project file. The bits on the right are merged better when I'm slicing it. I'm currently running master though so maybe that is something that was improved since we split off 4.2. We've been fixing small things as we built more integration tests in that area.

[image]

There are some bits that seem to protrude here, but in reality that is only because the line is rendered with an end cap which is not entirely true to reality. The coordinates of the line endpoints of the line can be seen by the ridge on the top side of the line. They seem to match exactly with the border of the model.