Ultimaker/Cura #6147: [4.2.1] Large "Minimum Support Area" and small "Support X/Y distance" causes supports to generate though smaller features of a model.

JohnEdwa, 2019-08-04:
**Application version** 
Cura 4.2.1

**Platform** 
Win 7 x64

**Printer** 
Creality Ender 3. [CE3_falcon_first_stagebottom.3mf.zip](https://github.com/Ultimaker/Cura/files/3464937/CE3_falcon_first_stagebottom.3mf.zip)

**Reproduction steps**
Slice the attached project/model. Only settings changed from stock are "Generate Supports: True" and "Support Placement: Everywhere".

**Actual results**
Supports will get generated through the smaller features of the model ignoring "Support X/Y Distance" unless it is increased enough, or if "Minimum Support Area", which is 10 by default in the new Creawesome settings, is reduced. I assume this is because Cura takes the X/Y distance into account when calculating if something is bigger or smaller than the "Minimum Support Area".

[image]

[image]
[image]

**Expected results**
Cura should't generate supports *through* the model, no matter how small they are.

Block137, 2019-08-04:
Same here
#5860

Ghostkeeper, 2019-08-11:
Does it help to reduce the Support Join Distance setting?

JohnEdwa, 2019-08-11:
Not really. 
It's 2mm by default, if you reduce it to around 0.5mm it starts noticing them but also creates supports with gaps and features that won't really print properly, and even at 0 it still won't notice [the smallest one in the middle once it starts going up. 
](https://user-images.githubusercontent.com/5417183/62837610-02c4e180-bc7a-11e9-9f4c-1c69785cfacb.PNG)

Ghostkeeper, 2019-08-12:
Yeah indeed the one in the middle seems really only to be affected by the Minimum Support Area for some reason.

Ellecross, 2019-08-14:
Hey @JohnEdwa looks like this has been happening even before 4.2.1. You see it now because the value of the profile settings have changed after the https://github.com/Ultimaker/Cura/pull/5809 was implemented. 

One of the settings that I noticed that changes this is the Minimum Support Area. It used to be 0.0 but after that PR 5809, it's now 10! 

In the old versions of Cura, with the same Creality Ender-3 printer and same model, I get the same behavior - the support crosses through the model. 

For now, you can change that setting back to the old default value. 

I'll bring this back to the Cura team for discussion.

Ellecross, 2019-08-15:
Hey @JohnEdwa, we've discussed this in the team and we think it needs more investigation. 

Cura developers, please see CURA-6724.

fvrmr, 2022-03-14:
Closing this issue since we couldn't reproduce this anymore.

closed 2022-03-14 by fvrmr