Ultimaker/Cura #6149: VAZE mode

Duha777, 2019-08-04:
Anycubic (D) PREDATOR Printer
Vase mode does not work (Spiral print external casing)
When using it, the model is printed rotated about the Y axis and the nozzle crashes into the table !!!!
On the preview of the G-code, everything is fine - but the printer does not understand this code ...
[image]
[D_ROCKET_V1_by_Spannerhands.3fm.zip](https://github.com/Ultimaker/Cura/files/3465028/D_ROCKET_V1_by_Spannerhands.3fm.zip)

Vandrasc, 2019-08-05:
Hi @Duha777 , which Cura version are you using? I just loaded the project file provided by you in the latest 4.2.1 and I cannot see the issue you encounter.  

Also, maybe you can try first to backup your configuration (Extensions>Cura Backups) and then simulate a clean install by removing your cura folder and retry.

Ghostkeeper, 2019-08-11:
He's using Cura 4.2.1, as can be seen thanks to the Version In Title plug-in from FieldOfView! :heart:

Ghostkeeper, 2019-08-11:
Maybe it's possible to indicate the part of the g-code where your printer goes wrong? The g-code looks fine on this end.

Duha777, 2019-08-11:
Yes, the G-code will look good - I checked it myself and did not find an error ...
But when using the "VASE" mode, the printer starts printing so that one of the three axes has incorrect commands ...
I understand that this is a printer firmware error - I just think that this can happen again on other Delta printers.
Without using the "VASE" mode, everything is OK.
Checked several times.
The mechanic of the printer is OK.
Using the same printer with a file cut in CURA version 4.0 is all OK.
The printer (or rather one of its axes) does not correctly understand the code received in CURA version 4.2.1.

Ghostkeeper, 2019-08-11:
Maybe the firmware is glitching out over the continuously increasing Z coordinate with spiralize mode?

If the g-code is correct though, I don't see that Cura could do anything to fix this for you. Maybe you need to take this up with the firmware developers of your printer.

Duha777, 2019-08-11:
The printer does not work correctly from the beginning of printing. The head rests on the bed after the first change in print direction.

Duha777, 2019-08-11:
Yes, I wrote to the printer manufacturer ... I'm waiting for an answer ...
This is not a problem for me - I just wanted this information to be available ... Maybe someone will have the same problems ...
  G - the code is correct - this is true! Checked!
This is a printer error ...
But why only in CYURA version 4.2.1?

Ghostkeeper, 2019-08-11:
Ah, so if it worked with Cura 4.1 then there must be some difference in the g-code between the two versions. The difference between the g-codes will be an indicator of what exactly triggers this bug.

Duha777, 2019-08-11:
I looked at both codes - but only the beginning of the print section ...
I'll look in more detail again.
Thanks for the advice :)

closed 2020-06-19 by no-response[bot]