Ultimaker/Cura #6151: Supports not being generated v4.2.1

CJinxed, 2019-08-04:
Models that require support structure are not having it generated by the program despite setting be on to generate them.

Previous version generated supports appropriately for the models, this doesn't happen since the update to this current version making me unable to print these models.

Vandrasc, 2019-08-05:
Hi @CJinxed , which printer definition are you using? You could use the bug template so we could receive more information in order to help us in reproducing this bug, or maybe you could share the project file with us? This way we could also test on the exactly the same configurations as you.

Thank you!

CJinxed, 2019-08-05:
All Events.stl
<https://drive.google.com/file/d/1_tbbPOcYgUf3iwdxKFTr92yiHNF7nMLO/view?usp=drive_web>
Hello

I am using a Creality Ender 3 Pro and trying to print the attached file,
these are minis from Thingverse, something called Pocket Tactics, saying
this as I have noticed the problem isn't limited to just these models,, the
supports just don't seem to generate as well anymore. It seems to ignore
something completely (like the overhanging axe, hand, shield and swords
through this file. I am using Standard Quality 0.2 Layer Height and 100%
Infill.

Is there a way I can roll back my version or Cura? as, right now, this
update is of 0 use to me and is more of a hindrance.

Thank You

Vandrasc, 2019-08-05:
Hi @CJinxed , this is because of the Creawesome mode introduced in 4.2 as the settings did change for the profiles.  We will have a look but I am afraid that this could be just the normal behavior after the creawesome implementation. 

There is a way to go back to your previous configuration if you didn't remove your previous cura directory. Remove the newest cura 4.2 folder and then just lunch the previous version. 

Help menu > configuration folder and you will see the current configuration which can be removed. 

Or, did you backup your old config. before the upgrade? Then you could just remove the entire cura folder and restore the backup from Extensions > Cura backups.

CJinxed, 2019-08-05:
Hello

Is it possible to redownload an older version of the software? I print mostly smaller things like that and if the ongoing versions will not generate the supports required I’m quite stuck; if I can download the older version I can keep using that and be fine. 

Thank You
Charles Jinkins

Vandrasc, 2019-08-05:
By default the creality profile sets the value to "touching buildplate" instead of "everywhere", could you try to change that and see if that fixes your problem? In the Support category, enable the support and set the Support Placement to Everywhere. 

You can find all Cura version here: https://ultimaker.com/software/ultimaker-cura.

Thank you both for your feedback!

CJinxed, 2019-08-05:
Hello

Unfortunately, in the case of models like that, it only starts doing the supports from the too of its base, not the bed, so still risks collapse. 

I found the older version though so will you that, thank you for your help.

Charles Jinkins

nallath, 2019-08-07:
The setting you are looking for is the minimum support area. If you set it to 0, the small supports will be generated again.

pfritzd, 2019-08-07:
I am running into the same issue.  Despite the generate support being "on", they are not showing in the Preview ... and therefore i assume that it has not been added to the Gcode.  Gcode attached
[ReckonHead.zip](https://github.com/Ultimaker/Cura/files/3477679/ReckonHead.zip)

Please advise.  In the meantime I will try to create my own supports and see if they render.

Thanks for the help

FromageTheDog, 2019-08-09:
Just to add another data point -- I am also have an Ender 3 printer and, since version 4.2.1, supports are not _consistently_ generated.  As a test, I've been trying to print parts from [this thing](https://www.thingiverse.com/thing:1327093).  Some elements will have supports generated (like the [Fan_half.zip](https://github.com/Ultimaker/Cura/files/3484187/Fan_half.zip)), while others trigger no supports at all, despite needing them (like the [HPC_Stage_1_No_Supports.zip](https://github.com/Ultimaker/Cura/files/3484185/HPC_Stage_1_No_Supports.zip)).

Ghostkeeper, 2019-08-11:
Closing this as duplicate of https://github.com/Ultimaker/Cura/issues/6134. Please continue the conversation there.

closed 2019-08-11 by Ghostkeeper