Ultimaker/Cura #6152: [4.2.1]  UpdateChecker crash cura on startup

bvalat, 2019-08-05:
**Application Version**
4.2.1

**Platform**
Win 10 64

**Printer**
Ender 3 (should not be relevant)

**Steps to Reproduce**
Install and try to start. 

**Actual Results**
Cura crash during version check.
Logs just shows the loading of the UpdateChecker plugin and stop here

**Expected results**
Cura starting.

**Additional Information**
I removed the UpdateChecker plugin from cura directory and cura starts.
I tried with Cura 4.1 UpdateChecker, it works as well.
I made a diff between the two version, seems like the only diff is the activation of https instead of http. So maybe a problem with TLS code?

diegopradogesto, 2019-08-05:
We noticed that for some people (mainly using macOS or Linux) the update checker doesn't work because of the SSL certificates. However we tested on Windows and it works and it should never crash.

We made a fix for it but is not release yet. If we provide you a test build, could you check for us if it works and it doesn't crash?

LipuFei, 2019-08-05:
Hi @bvalat , we have a master build with a potential fix for this. Could you try it for us? Be aware that although its version number is 4.1.2, it's actually built with the current `master` branch. It is advised to backup your `cura\4.1` directory first if you have one.

https://we.tl/t-nI0EAND9Pw

bvalat, 2019-08-05:
Your version crash as well.
I remove the s of https and still crashing.
But if i replace updatechecker plugin by the 4.1 one it works.

Ghostkeeper, 2019-08-06:
[This](https://github.com/Ultimaker/Uranium/commit/341937cc876eabada25fbe0674b90e443e59f629) and [this](https://github.com/Ultimaker/Uranium/commit/b0d0f30eb04efe071599ecf566616fc0611e7109) are the only commits made between 4.1 and 4.2. The second one there only changes some code style really, so I'd expect that replacing the plug-in of 4.2 with the version from 4.1 would result in the exact same files as removing the "s" from https.

Routout, 2019-08-08:
I can confirm this also. Windows 10. Cura wont start at all. When update checker is removed it starts.

LipuFei, 2019-08-08:
JIRA ticket: CURA-6698

SkiddieTech, 2019-08-10:
I can also confirm this issue W10 64bit, however it still wont start when "UpdateChecker" is removed from plugins. 4.1.0 works fine

bvalat, 2019-08-11:
In my case it seems that even if it starts, i cannot save a gcode....
I had to disable "anonymous stats".
Since i never manage to sign in my utlimaker account, it seems that anything related to communicate with ultimaker website is craching cura for me : Signing in, anonymous stats, update checker. 
If manage to run cura from code, evrything worked fine.
I reinstalled windows from scratch (i formated my system disk) and stil  have the problem.
Maybe if you can find something common between those 3 features, we'll find the problem.

LipuFei, 2019-08-12:
Hi there, we cannot reproduce this problem on our Windows machines... Could you run "System Information" in the "Start" menu and show us your Windows versions? That may help. And any additional information would help, too. Thanks!

Here a screenshot of my "System Information":
[image]

Routout, 2019-08-12:
I have 10 64 Pro
Version	10.0.18362 Build 18362

Today again cura didn't start. I had to shut cura.exe mutiple times fron the task manager before it started. I did remove that update checker already earlier from plugins folder, so theres something else than that causing this.
I couldn't find any logs from event viewer regarding this.
Does Cura have any logs which could help?

LipuFei, 2019-08-12:
Hi @Routout, it's the current Insider Preview version, right? 18362. @bvalat @SkiddieTech Are you also using this preview version?

Routout, 2019-08-12:
I checked that I'm on normal update channel, not insider preview.

bvalat, 2019-08-12:
No i'm on standard build (1903 may update). 
I did try yesterday the last insider build with no success so i reinstalled windows from scratch (partition formated).
See https://en.wikipedia.org/wiki/Windows_10_version_history (1903 is 18362 build).
I have another computer and everything seems OK. Will compare them.
The non working is a beelink m1 N3450 with 6gb ram intel graphic card with win 10 familly in french 1903 fresh install from yesterday.
The working one is a selfbuild I7 2600K 24Go ram nvidia gtx1070 with win10 professional in french (upgraded from familly edition). Build should be 1903 as well. Will confirm tonight.

LipuFei, 2019-08-12:
I just tried on my laptop with 18362 and it seems OK.

bvalat, 2019-08-12:
@Routout :
Since i reinstalled my computer, i have same behaviour.
I must remove updatechecker to start Cura but some times i have to start it several times before getting cura start.
The non starting process seems to hang before cura loading screen and stays aroud 53Mo if i remember.

LipuFei, 2019-08-12:
> 
> 
> I checked that I'm on normal update channel, not insider preview.

You are right. 18362 is from the normal channel. For some reason I have this laptop doesn't agree on that.

bvalat, 2019-08-12:
@everybody : Is your windows in english or native language  (for me in french).
@Lipufei : It was the "gold" prerelease build as well. so it went from fast to slow ring and then  to main release.

Routout, 2019-08-12:
Same behavour here also  @bvalat about 53M.
I have English language, but finnish formats.

bvalat, 2019-08-12:
Maybe we should try to start Cura without network just to see if it's network related. (I'm at work, won't be able to test before 4 hours).
I feel that is a network layer (SSL/Socket/....) conflict with something on some windows. (language, local,...)

LipuFei, 2019-08-12:
Mine are on English with Dutch formats.

> 
> 
> Maybe we should try to start Cura without network just to see if it's network related. (I'm at work, won't be able to test before 4 hours).
> I feel that is a network layer (SSL/Socket/....) conflict with something on some windows. (language, local,...)

I agree. Just for more info, the code that's crashing uses `urllib` from python to issue HTTP requests. Our python version 3.5.2 natively compiled on our build VM, and it's linked statically against OpenSSL 1.0.2h.

@Routout @bvalat @SkiddieTech  Could you try something for us?

1. Download [python-3.5.2-embed-amd64.zip](https://www.python.org/ftp/python/3.5.2/python-3.5.2-embed-amd64.zip) from https://www.python.org/downloads/release/python-352/
2. Extract it to a folder run the `python.exe` in that folder.
3. Run the following command and see if you get a crash:
```
Python 3.5.2 (v3.5.2:4def2a2901a5, Jun 25 2016, 22:18:55) [MSC v.1900 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import urllib.request
>>> f = urllib.request.urlopen("https://software.ultimaker.com/latest.json")
>>> print(f.read().decode('utf-8'))
```

```

Routout, 2019-08-12:
Seem to work, but also cura works atm...

```
Python 3.5.2 (v3.5.2:4def2a2901a5, Jun 25 2016, 22:18:55) [MSC v.1900 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import urllib.request
>>> f = urllib.request.urlopen("https://software.ultimaker.com/latest.json")
>>> print(f.read().decode('utf-8'))
{
    "cura": {
        "Windows": {
            "major": 4,
            "minor": 2,
            "revision": 1,
            "url": "https://ultimaker.com/en/products/ultimaker-cura-software?utm_source=cura&utm_medium=software&utm_campaign=sw-update"
        },
        "Linux": {
            "major": 4,
            "minor": 2,
            "revision": 1,
            "url": "https://ultimaker.com/en/products/ultimaker-cura-software?utm_source=cura&utm_medium=software&utm_campaign=sw-update"
        },
        "Darwin": {
            "major": 4,
            "minor": 2,
            "revision": 1,
            "url": "https://ultimaker.com/en/products/ultimaker-cura-software?utm_source=cura&utm_medium=software&utm_campaign=sw-update"
        }
    }
}

```

LipuFei, 2019-08-12:
> Seem to work, but also cura works atm...

@Routout So for you it sometimes works sometimes doesn't?

LipuFei, 2019-08-12:
I'm not sure but this Python issue seems related: https://bugs.python.org/issue36011

Also partially explains behaviors like once it works, it just works...

I'll do some testing to verify this.

Routout, 2019-08-12:
> > Seem to work, but also cura works atm...
> 
> @Routout So for you it sometimes works sometimes doesn't?

Yes. I have sometimes kill process manually several times before Cura starts. Then it might startup properly a few times.

LipuFei, 2019-08-12:
Please correct me if I'm wrong, but I think it's this Python issue https://bugs.python.org/issue36011

**TL;DR:** One way to see if it's issue 36011 is to access https://software.ultimaker.com via IE/Edge just once. Then open Cura without any modification. If it doesn't crash on that CERTIFICATE_VERIFY_FAILED error, it's very likely caused by issue 36011.

**Details:**
I tested on a newly created Windows Server 2019 VM:
 - Download [python-3.5.2-embed-amd64.zip](https://www.python.org/ftp/python/3.5.2/python-3.5.2-embed-amd64.zip) from https://www.python.org/downloads/release/python-352/
 - Extract to a folder (say py35) and run the `python.exe` file in that folder
 - Now you get your python console. Run the following commands:

```
> import urllib.request
> f = urllib.request.urlopen("https://software.ultimaker.com/latest.json")
...
ssl.SSLError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed
>
> import ssl
> c = ssl.SSLContext(ssl.PROTOCOL_TLSv1_2)
> len(c.get_ca_certs())
0
> c.load_default_certs()
> len(c.get_ca_certs())
13   # I have 13 default CA certificates
```

 - So I have 13 default CA certificates that are trusted.
 - Now I open IE/Edge and open the webpage https://software.ultimaker.com
 - Run the python command to access https://software.ultimaker.com/latest.json again

```
> import urllib.request
> f = urllib.request.urlopen("https://software.ultimaker.com/latest.json")
# It works now!

> import ssl
> c = ssl.SSLContext(ssl.PROTOCOL_TLSv1_2)
> len(c.get_ca_certs())
0
> c.load_default_certs()
> len(c.get_ca_certs())
16   # Now I have 16 default CA certificates
```
 - So after I have accessed https://software.ultimaker.com via IE/Edge, I got 3 more trusted CA certificates and the `urlopen()` request now works.

The problem is that Windows 10 comes with a minimal number of trusted CA certificates in its default certificate store. The SSL certificate of https://software.ultimaker.com is issued by Let's Encrypt, whose CA certificate is not in the default certificate store. After you use IE/Edge to access https://software.ultimaker.com, it uses SChannel which magically fetches and stores new CA certificates into the default CA store on Windows. After this, if you try to access https://software.ultimaker.com in Python, it works.

LipuFei, 2019-08-12:
@bvalat @SkiddieTech @Routout

Routout, 2019-08-12:
No errors for me. But again Cura wont start...

```
Python 3.5.2 (v3.5.2:4def2a2901a5, Jun 25 2016, 22:18:55) [MSC v.1900 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import urllib.request
>>> f = urllib.request.urlopen("https://software.ultimaker.com/latest.json")
>>> import ssl
>>> c = ssl.SSLContext(ssl.PROTOCOL_TLSv1_2)
>>> len(c.get_ca_certs())
0
>>> c.load_default_certs()
>>> len(c.get_ca_certs())
67
>>>
```

bvalat, 2019-08-12:
Maybe you have found a lead for me.
Python 3.5.2 is crahsing silently but not 3.5.4 (And when i ran it from code it was with 3.5.4).
Here is 3.5.2  :
```
C:\py>python.exe
Python 3.5.2 (v3.5.2:4def2a2901a5, Jun 25 2016, 22:18:55) [MSC v.1900 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import urllib.request
>>> f = urllib.request.urlopen("https://software.ultimaker.com/latest.json")

C:\py>
```
```
C:\py>python.exe
Python 3.5.2 (v3.5.2:4def2a2901a5, Jun 25 2016, 22:18:55) [MSC v.1900 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import urllib.request
>>> f = urllib.request.urlopen("https://software.ultimaker.com/latest.json")

C:\py>
```

bvalat, 2019-08-12:
YES!!!!!!
I replaced _ssl.pyd by the one of 3.5.4 [https://www.python.org/ftp/python/3.5.4/python-3.5.4-embed-amd64.zip](url) and everything works for me. Even my old unresolved #5817

LipuFei, 2019-08-12:
> No errors for me. But again Cura wont start...
@Routout Could you upload your log file? You can find it in `C:\Users\<your-user-name>\AppData\Roaming\cura\4.2\`

Ghostkeeper, 2019-08-12:
There can hardly be a safer time to update our dependencies than this, with the last release just out the next release is far away.

LipuFei, 2019-08-13:
> 
> 
> YES!!!!!!
> I replaced _ssl.pyd by the one of 3.5.4 [https://www.python.org/ftp/python/3.5.4/python-3.5.4-embed-amd64.zip](url) and everything works for me. Even my old unresolved #5817
> 
> Maybe it's time to update python version embedded in cura?
> yes i know, easier to say, harder to do but i think a least 3.5.4 should be not too painful.

@bvalat Glad that you found a solution! Thank you very much for the information!

I compared the bundled OpenSSL versions in Python 3.5 for Windows:

 - Python 3.5.2: openssl-1.0.2h
 - Python 3.5.4 to 3.5.7: openssl-1.0.2k

There's indeed a difference in which OpenSSL they use.

I also tested both python 3.5.2 and 3.5.4 with the embedded ZIPs from python.org on a clean Windows Server 2019. They both give the `CERTIFICATE_VERIFY_FAILED` error message in my case. You can find an example error below. So, I think your silent crash is something related to the OpenSSL version that's being used, but not the CA certificates. As described in [python issue 36011](https://bugs.python.org/issue36011), if you get the CA certificate issue once, you will keep getting it until the necessary CA certificates are imported into the Windows certificate store.

```
>>> import urllib.request
>>> f = urllib.request.urlopen("https://software.ultimaker.com/latest.json")
Traceback (most recent call last):
  File "urllib\request.py", line 1254, in do_open
  File "http\client.py", line 1107, in request
  File "http\client.py", line 1152, in _send_request
  File "http\client.py", line 1103, in endheaders
  File "http\client.py", line 934, in _send_output
  File "http\client.py", line 877, in send
  File "http\client.py", line 1261, in connect
  File "ssl.py", line 385, in wrap_socket
  File "ssl.py", line 760, in __init__
  File "ssl.py", line 996, in do_handshake
  File "ssl.py", line 641, in do_handshake
ssl.SSLError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:719)
```

We will discuss the Python upgrade. If I remember correct, upgrading to 3.5.4+ was not so easy with our current build system... 3.5.4+ requires VS2017 while we only have VS2015 in the system. It will take more effort than just changing the version we use.

Routout, 2019-08-13:
> > No errors for me. But again Cura wont start...
> 
> @Routout Could you upload your log file? You can find it in `C:\Users\<your-user-name>\AppData\Roaming\cura\4.2\`

```
2019-08-13 10:58:02,484 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin FileLogger
2019-08-13 10:58:02,499 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin FirmwareUpdateChecker
2019-08-13 10:58:02,511 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin FirmwareUpdater
2019-08-13 10:58:02,522 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin GCodeGzReader
2019-08-13 10:58:02,534 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin GCodeGzWriter
2019-08-13 10:58:02,545 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin GCodeProfileReader
2019-08-13 10:58:02,567 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin GCodeReader
2019-08-13 10:58:02,580 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin GCodeWriter
2019-08-13 10:58:02,595 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin ImageReader
2019-08-13 10:58:02,607 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin LegacyProfileReader
2019-08-13 10:58:02,620 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin LocalContainerProvider
2019-08-13 10:58:02,634 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin LocalFileOutputDevice
2019-08-13 10:58:02,645 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin MachineSettingsAction
2019-08-13 10:58:02,657 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin ModelChecker
2019-08-13 10:58:02,670 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin MonitorStage
2019-08-13 10:58:02,683 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin PerObjectSettingsTool
2019-08-13 10:58:02,696 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin PostProcessingPlugin
2019-08-13 10:58:02,708 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin PrepareStage
2019-08-13 10:58:02,720 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin PreviewStage
2019-08-13 10:58:02,742 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin RemovableDriveOutputDevice
2019-08-13 10:58:02,766 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SimulationView
2019-08-13 10:58:02,783 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SliceInfoPlugin
2019-08-13 10:58:02,797 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SolidView
2019-08-13 10:58:02,811 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SupportEraser
2019-08-13 10:58:02,834 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService.getAccessToken [112]: No auth data to retrieve the access_token from
2019-08-13 10:58:02,838 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin Toolbox
2019-08-13 10:58:02,853 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin CameraTool
2019-08-13 10:58:02,869 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin MirrorTool
2019-08-13 10:58:02,886 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin RotateTool
2019-08-13 10:58:02,903 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin ScaleTool
2019-08-13 10:58:02,915 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SelectionTool
2019-08-13 10:58:02,933 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin TranslateTool
2019-08-13 10:58:02,979 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin UFPReader
2019-08-13 10:58:02,994 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin UFPWriter
2019-08-13 10:58:03,011 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin UltimakerMachineActions
2019-08-13 10:58:03,121 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin UM3NetworkPrinting
2019-08-13 10:58:03,143 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin UpdateChecker
2019-08-13 10:58:03,146 - INFO - [Thread-1] UpdateChecker.UpdateCheckerJob.run [36]: Checking for new version of cura
2019-08-13 10:58:03,486 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin USBPrinting
2019-08-13 10:58:03,654 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade21to22
2019-08-13 10:58:03,722 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade22to24
2019-08-13 10:58:03,790 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade25to26
2019-08-13 10:58:03,866 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade26to27
2019-08-13 10:58:03,941 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade27to30
2019-08-13 10:58:04,015 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade30to31
2019-08-13 10:58:04,080 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade32to33
2019-08-13 10:58:04,128 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade33to34
[... cut ...]
2019-08-13 10:58:20,017 - DEBUG - [MainThread] UM.Qt.QtApplication.windowClosed [434]: Shutting down cura
2019-08-13 10:58:20,117 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [292]: Attempting to kill the engine process
2019-08-13 10:58:20,118 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [298]: Killing engine process
2019-08-13 10:58:20,122 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [301]: Engine process is killed. Received return code 1
```

Routout, 2019-08-13:
Earlier is after normally started and closed cura. After that I did a windows reboot and again Cura crashes, and theres nothing in the logs.

LipuFei, 2019-08-13:
Hi @bvalat @Routout @SkiddieTech , I've created a Cura build with:
 - Python 3.5.7
 - A fix for trusted CA certificates (https://github.com/Ultimaker/Uranium/pull/507)

The download link is https://we.tl/t-SdbysJKPzg. Could you help us to test it? Thanks!

Please note that although this build has a file name Cura-4.1.99, it's actually created with the latest changes from the `master` branch, so this build is not so stable, it's only for testing purposes. Also backup your `C:\Users\<your-user-name>\AppData\Roaming\cura\4.1` if you have one, because this build will use that folder.

bvalat, 2019-08-13:
Your test Version is working for me. No more updatechecker plugin crash or file saving with anonymous data enabled and i can sign in my cloud account.
So it fixes this ticket.
But it still has the "won't start sometimes". I have to start 3 times. The two first launch just hang (0kb stderr stdout). I think this is another issue so i'll create another issue for it.

LipuFei, 2019-08-13:
@bvalat Many thanks! I think the OpenSSL issues we found so far should be fixed. Note that this build is not really stable, so it's hard to say if it's still a problem with the HTTPS connections. We can check the log files see what's going on.

bvalat, 2019-08-13:
The problem is that when it hangs, there is no log or starting screen.
Just the cura process go up to ~60Mb and then fall basck to ~54-53mb and stays that way.
But if i launch it several times it finishes to launch.

bvalat, 2019-08-13:
I forgot to thank you for a such quick 3.5.7 build.
Your test build and official 4.2.1 behave the same way for the hang on start.
Since we had those problem before upgrading ssl and/or python, i'm not sure it's realy related so I opened another issue for the hang part.

LipuFei, 2019-08-13:
@bvalat I will wrap this ticket up. Let's discuss the hanging issue in #6197.

closed 2019-08-13 by LipuFei