Ultimaker/Cura #6154: [4.2.1] [4.0] AutoDetectBaudJob.py under linux fails to connect to SKR mini e3

tomoinn, 2019-08-05:
**Application version**
4.2.1 and 4.0, other versions not tested

**Platform**
Linux mint 19.2 - output of `uname -a` as follows:

```
Linux Ogre 5.0.0-23-generic #24~18.04.1-Ubuntu SMP Mon Jul 29 16:12:28 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```

**Printer**
Ender 3 with SKR mini E3 mainboard

**Reproduction steps**
Run Cura and then connect USB cable to printer, or run Cura after USB cable is already connected.

**Actual results**
Printer display freezes, printer is not found and displayed in monitor view. Logs show:

```
2019-08-05 16:33:24,817 - DEBUG - [MainThread] cura.TaskManagement.OnExitCallbackManager.addCallback [28]: on-app-exit callback [<bound method USBPrinterOutputDevice._checkActivePrintingUponAppExit of <USBPrinting.USBPrinterOutputDevice.USBPrinterOutputDevice object at 0x7f4d3001b948>>] added.
2019-08-05 16:33:24,819 - DEBUG - [Thread-2] USBPrinting.AutoDetectBaudJob.run [24]: Auto detect baud rate started.
2019-08-05 16:33:31,365 - DEBUG - [Thread-2] USBPrinting.AutoDetectBaudJob.run [50]: Checking /dev/ttyACM0 if baud rate 115200 works. Retry nr: 0. Wait timeout: 3
2019-08-05 16:33:31,372 - WARNING - [Thread-2] UM.Logger.logException [84]: Exception: Unable to create serial
2019-08-05 16:33:31,375 - WARNING - [Thread-2] UM.Logger.logException [88]: Traceback (most recent call last):
2019-08-05 16:33:31,378 - WARNING - [Thread-2] UM.Logger.logException [88]:   File "/build/env/4.2/inst/lib/python3.5/site-packages/serial/serialposix.py", line 265, in open
2019-08-05 16:33:31,380 - WARNING - [Thread-2] UM.Logger.logException [88]: OSError: [Errno 16] Device or resource busy: '/dev/ttyACM0'
2019-08-05 16:33:31,382 - WARNING - [Thread-2] UM.Logger.logException [88]: 
2019-08-05 16:33:31,384 - WARNING - [Thread-2] UM.Logger.logException [88]: During handling of the above exception, another exception occurred:
2019-08-05 16:33:31,387 - WARNING - [Thread-2] UM.Logger.logException [88]: 
2019-08-05 16:33:31,389 - WARNING - [Thread-2] UM.Logger.logException [88]: Traceback (most recent call last):
2019-08-05 16:33:31,391 - WARNING - [Thread-2] UM.Logger.logException [88]:   File "/tmp/.mount_UltimaV5gknm/usr/bin/plugins/plugins/USBPrinting/AutoDetectBaudJob.py", line 54, in run
2019-08-05 16:33:31,393 - WARNING - [Thread-2] UM.Logger.logException [88]:     serial = Serial(str(self._serial_port), baud_rate, timeout = read_timeout, writeTimeout = write_timeout)
2019-08-05 16:33:31,395 - WARNING - [Thread-2] UM.Logger.logException [88]:   File "/build/env/4.2/inst/lib/python3.5/site-packages/serial/serialutil.py", line 240, in __init__
2019-08-05 16:33:31,398 - WARNING - [Thread-2] UM.Logger.logException [88]:   File "/build/env/4.2/inst/lib/python3.5/site-packages/serial/serialposix.py", line 268, in open
2019-08-05 16:33:31,402 - WARNING - [Thread-2] UM.Logger.logException [88]: serial.serialutil.SerialException: [Errno 16] could not open port /dev/ttyACM0: [Errno 16] Device or resource busy: '/dev/ttyACM0'
```

**Expected results**
Printer connection is shown in monitor view

**Additional information**
This is something to do with the use of the ISP python library in AutoDetectBaudJob.py. I suspect it's claiming the serial port and then not releasing it correctly, at which point the attempt to create a Serial object fails with the error shown above. Commenting out the lines which refer to this library gets rid of this problem entirely and detection works immediately:

```python
 def run(self) -> None:
        Logger.log("d", "Auto detect baud rate started.")
        wait_response_timeouts = [5, 15, 30]
        wait_bootloader_times = [5, 15, 30]
        write_timeout = 3
        read_timeout = 3
        tries = 2

        # programmer = Stk500v2()
        serial = None
        # try:
        #     programmer.connect(self._serial_port)
        #     serial = programmer.leaveISP()
        # except ispBase.IspError:
        #     programmer.close()
```

The corresponding log snippet is then:

```
2019-08-05 18:00:05,358 - DEBUG - [Thread-2] USBPrinting.AutoDetectBaudJob.run [24]: Auto detect baud rate started.
2019-08-05 18:00:05,358 - DEBUG - [Thread-2] USBPrinting.AutoDetectBaudJob.run [50]: Checking /dev/ttyACM0 if baud rate 115200 works. Retry nr: 0. Wait timeout: 5
2019-08-05 18:00:06,564 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [188]: Error check finished, result = False, time = 3.7s
2019-08-05 18:00:10,367 - DEBUG - [Thread-2] USBPrinting.AutoDetectBaudJob.run [82]: Detected baud rate 115200 on serial /dev/ttyACM0 on retry 0 with after 0.00 seconds.
2019-08-05 18:00:13,404 - DEBUG - [MainThread] cura.AutoSave._onTimeout [56]: Autosaving preferences, instances and profiles
2019-08-05 18:00:13,422 - INFO - [Thread-13] USBPrinting.USBPrinterOutputDevice._setFirmwareName [327]: Unknown USB output device Firmware name: b'FIRMWARE_NAME:Marlin bugfix-2.0.x (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:SKR-Mini-E3 EXTRUDER_COUNT:1 UUID:02aaf9c2-aea9-4a2e-aa54-2b798bc0830e\n'
```

rburema, 2019-08-08:
Hi @tomoinn, Thanks for the detailed report. I'll discuss it with my colleages, but please keep in mind we can't really test very well for printers other than our own.

rburema, 2019-08-09:
Sorry, like I said, we can't test/fix this for printers we don't have at the office.

(By the way are you using the AppImage?)

closed 2019-08-09 by rburema