Ultimaker/Cura #6156: Add "Material" settings to Z-Height or Mesh Modifiers

AbeFM, 2019-08-05:
**Is your feature request related to a problem? Please describe.**
Yes.
I would like a better ability to control print parameters to develop print profiles. Things like "extra prime after restart" have to be done as a series of prints.

**Describe the solution you'd like**
I would like to include more of the Material Settings to the types of changes which can be made per model, per height or per modifier.

**Describe alternatives you've considered**
I've thought about generating g-code, and splicing it together at by taking the next XXX lines and pasting it into one file to generate the behavior I will need. I'm in fact going to be doing that after I write this post. :-)

**Affected users and/or printers**
People who want to fine tune/rough tune their profiles.

**Additional context**
I imagine this is harder to do, maybe it's a CURA Engine issue?
[image]
Only one option under Material at all.

Vandrasc, 2019-08-06:
Hi @AbeFM , did you checked the "Show all"? 
I do get more settings by enabling the "Show all" settings:
[image]

Also, did you checked the Material settings plugin (Marketplace)?

AbeFM, 2019-08-06:
Thanks for the reminders! You caught me a little off guard since I hadn't added new plug ins yet.

So I did, and the issue remains - I get the same settings you do, flow, flow and flow. But no retraction, k-factor, extra restart distance, temperature, or any of these:

[image]

Ghostkeeper, 2019-08-11:
There is a good reason for all of these though:
* All of the temperature settings are regulated in your printer by a PID regulator, and it needs some time to stabilise the temperature. Changing the temperature while it's stabilising tends to throw off these regulators causing it to fluctuate wildly. Switching temperature at every mesh is the perfect storm for this phenomenon. We've seen the temperature fluctuating by 80 degrees up and down. That basically burns your PLA to a crisp.
* It makes no sense to claim your nozzle cools down faster while printing one object than while printing another.
* Adhesion tendency is used to determine which material gets put outside in the prime tower, which can't be different for each model since there is only one prime tower. It's meant to describe an inherent property of the material anyway.
* Surface energy is used only to determine support settings for Breakaway profiles at the moment. It's meant to describe an inherent property of the material anyway.
* Shrinkage ratio, crystalline material, anti-ooze retracted position, anti-ooze retraction speed, break preparation retracted position, break preparation retraction speed, break retracted position, break retraction speed and break temperature are not used by Cura but used by firmware that is able to read the [material profiles](https://github.com/Ultimaker/fdm_materials).
* Retractions are made between objects, and so don't belong to any particular object. Similar to how the travel speeds can't be adjusted per object, you also can't adjust retractions. You could argue that travel moves within one object (like between infill lines) should be adjustable, but our engine's architecture currently doesn't keep track of that as we only store one speed, retraction distance, etc. per line config, which is the line settings that infill/skin/wall lines of an object should be drawn with. We don't see a big enough use case to warrant implementing this right now.
* Maximum Retraction Count and Minimum Extrusion Distance Window work on the length of filament that a feeder extrudes. Since there is only one feeder per extruder and not one per object in your printer, it also can't be set per object.
* Nozzle switching settings can't be adjusted per object since there is less than one switch per layer per extruder, not one per object (and they'd be between objects anyway).
* Linear Advance Factor is not a Cura setting, but the result of [this plug-in](https://github.com/fieldOfView/Cura-LinearAdvanceSettingPlugin). Presumably the setting can't be changed per object because the plug-in was made before Cura put any comments in the g-code as to which mesh is being printed; and it'd be pretty hard to match that comment with a certain mesh in the scene anyway.

In short, we're going to reject this because there is nothing we can feasibly improve this.

closed 2019-08-11 by Ghostkeeper

AbeFM, 2019-08-12:
AH! I made a mistake there, I didn't check specific items in my screen shot, they just happened to be checked. I appreciate the detailed responses and I apologize for leading you to take the time. Two main points:

1) Many of these would be just as useful if I could switch them per-layer, so anything that's hard to do per object or per modifier mesh doesn't need to be done there.
2) The things I care about are limited to testing materials:

Per Object:
   Retraction: distance, speed, extra prime, 

Per Layer:
   Nozzle Switch: Speed, distance, prime, 
   Temperature: Printing/Default/Starting/Final

I don't care, but I could see why people would want:
Anti-Ooze positions, etc, to keep travels small/predictable

These will all shave a LOT of time off getting a printer/material tuned. Trying to do a temperature tower now is really hard because I have to tweak starting, final, printing temps for each extruder, as a function of height. It's a lot of notepad cut-and-paste and really takes some time and is error prone.

Here's my other comments, likely less important:

> There is a good reason for all of these though:
I assumed there were good reasons for most of it, as you've outlined many of them are.

However, I would reiterate that some of these are super useful to the end user, and may be worth a second look - specifically:

> * All of the temperature settings are regulated in your printer by a PID regulator, and it needs some time to stabilise the temperature. Changing the temperature while it's stabilising tends to [...]

Even though it will not regularly be useful for doing routine printing of objects you want at different temperatures (unwise for the reasons you mentioned), it can still be super useful for developing the right temperature to test at.
*Especially* when my temperatures are different between nozzles, it can be quite tedious to hand edit files to do a temperature tower.

> * It makes no sense to claim your nozzle cools down faster while printing one object than while printing another.

It certainly does - one nozzle printing with a different size, line width, etc, WILL cool at a different rate because it's melting a different amount of plastic. In some (my) cases, the heat blocks, nozzles, heaters, and everything else is different. This is not a high priority feature for me, though, but it is incorrect to say they don't cool at different rates.

Ah, I've realized what happened, it's my fault, sorry, I'll only respond to what makes sense:

> * Retractions are made between objects, and so don't belong to any particular object. Similar to how the travel speeds can't be adjusted per object, you also can't adjust retractions. You could [...]
This is totally worth putting in, again, for tuning, not for printing. I would just retract based on the object you're leaving and not worry about it? I guess there's an un-retract amount to keep track of during the switch, but even if the first unretract was wrong I wouldn't care.

I envision temperature towers side by side. 
[image]

By changing the retract as I left each object, I would hope to see changes in stringing between the towers. And if I'm doing temperature changes with height, I'd have a really useful test here. 

> * Nozzle switching settings can't be adjusted per object since there is less than one switch per layer per extruder, not one per object (and they'd be between objects anyway).
Agreed - this would clearly only be every layer or on occasional layers.

> * Linear Advance Factor is not a Cura setting, but the result of [this plug-in](https://github.com/fieldOfView/Cura-LinearAdvanceSettingPlugin). Presumably the setting can't be changed per object [...]

`G1 X145.881 Y141.088 E19.31946
G0 F2700 X145.276 Y140.842
G1 F2400 E18.51946
;MESH:Object 2
G0 F3600 X145.915 Y139.716
G0 X150.242 Y139.814
G0 X151.43 Y141.707
G0 X153.991 Y141.819`
It seems like you know what mesh you're working on? Or you mean, for overlap/modifier volumes. Probably that. I would be WELL happy if it was something that just applied per object, it would already help a lot! Just changing:

;MESH:Object 2

to

;MESH:Object 2
M900 K{K_ObjectN}; sets k-factor for ObjectN

might work?

> We once had these settings enabled if you're printing in one-at-a-time mode for a while by setting them as per-mesh-group settings which is in CuraEngine interpreted as almost global settings. This could be done because for one-at-a-time mode CuraEngine almost completely restarts the slice for every object. However then we discovered that the settings never had any effect on the print at all because they were always being overwritten by per-extruder settings. So we removed that exception. Nobody ever noticed that they were added or removed as far as we know. Our statistics don't show any significant >0.01% of users using it. One-at-a-time mode in general doesn't get the attention it deserves.

I played with it once, didn't see any changes happening, and stopped using it. It's one of the few things I actually miss from other slicers, CURA really handles nearly everything better than the other guys.

> In short, we're going to reject this because there is nothing we can feasibly improve on this.
Made good sense the way I presented it. I argue that it's worth doing if done right.

fieldOfView, 2019-08-13:
> Just changing:
> ```
>;MESH:Object 2
> ```
>to
>```
>;MESH:Object 2
>M900 K{K_ObjectN}; sets k-factor for ObjectN
>```
>might work?

That would have to be a feature request for the Cura-LinearAdvancePlugin (https://github.com/fieldOfView/Cura-LinearAdvanceSettingPlugin). Be prepared to write a very good use-case for it though ("it would be very useful for me" is not a usecase), because it will be significant work to get it to work properly.

AbeFM, 2019-08-13:
The use case is for testing. :-)
Right now I sneak M900's into my extruder start code, with significantly different extruders I need it. The reason I brought it up was for rapid testing.

So the question becomes - can this be supported at some reasonable level of effort?

I do not know, but I believe switching this as a function of layer (ideally, layer + extruder), would make this easier to do and still open up the possibilities of k-factor towers.

[image]

If you think there's enough merit there to at least make my case, I will.

Ghostkeeper, 2019-08-14:
> Even though it will not regularly be useful for doing routine printing of objects you want at different temperatures (unwise for the reasons you mentioned), it can still be super useful for developing the right temperature to test at.

I don't think it'll be a reliable test, for the same reasons.

> It seems like you know what mesh you're working on? Or you mean, for overlap/modifier volumes. Probably that. I would be WELL happy if it was something that just applied per object, it would already help a lot! Just changing:

Yeah since 4.0 it's possible to do this. However it'll be very hard to match the object in the scene (which has the per-object settings) with a mesh from the g-code. In the g-code it just puts the filename that the mesh was loaded from, but this is unreliable if:

* You duplicated meshes.
* You simply loaded the same file twice.
* You loaded a 3MF file which contains multiple meshes.
* You created a mesh in the scene directly with a plug-in, such as what the Support Blocker does with its cubes.

There's a workaround in that you can replace AbeFM's suggestion in the Search-and-Replace post-processing script. It'll have the same problems, though.

Switching settings per layer is something we've wanted for a long time. In our architecture change in CuraEngine last autumn we've paved some of the way to do that in the future, but it's not there yet by a long shot.

AbeFM, 2019-08-14:
Ah, ok. Well, since "do it by layer" was my hope, and "do it arbitrarily" was my moonshot, it does sound like it's "in process... but don't hold  your breath".

I can't do much beyond provide stunning ideas, and if you've already had them then all I can do is say I agree. It would be awesome to tune some of the settings, post processing is how I do it now.

**Is there a script or two I can look at, modify and test myself?** I could try to write something to pick out the temperatures, at least. Is there some way to express the "initial", "final", and "standby" print temperatures as deltas from the default? I can over-ride them to specific values but then I (think I) lose the 'calculate' feature.

That might still be hard to do with post processing if one nozzle's standby temp is the same as another's final temp, for example.

Again, thanks for the considered reply.

Ghostkeeper, 2019-08-15:
> I can't do much beyond provide stunning ideas, and if you've already had them then all I can do is say I agree.

Honestly in the Cura team you are known as a person that gives a lot of well-thought-out feedback. I hate that I have to shoot down so many of them because of time constraints and because the solutions are often complex.

> **Is there a script or two I can look at, modify and test myself?**

I think the easiest example script to start learning how to write post-processing scripts is the search-and-replace script. It's a really straightforward script. The gist of post-processing scripts is this: You provide a function that returns a dictionary of settings for the script (in a certain format). And you provide a function that takes a list of strings, one for each layer, and then returns a list of transformed strings, one for each layer.

> Is there some way to express the "initial", "final", and "standby" print temperatures as deltas from the default? I can over-ride them to specific values but then I (think I) lose the 'calculate' feature.

Yes, there is, but not from the GUI. It's reasonably simple to do for someone who dabbles in programming. First you create a profile, then go to Help -> Show Configuration Folder and close Cura. In the folder that popped up, go to the `quality_changes` subfolder and find a file there that looks like the name you gave your custom profile. There will be multiple: one for global settings and one for each extruder. Open the extruder profile with a text editor. Underneath the `[values]` header, you should add something like this:

    material_final_print_temperature = =material_print_temperature + 5
    material_initial_print_temperature = =material_print_temperature - 5

Adjust the formulas as you see fit. Note the double `=` though; the second `=` indicates that a Python expression follows. The values for all of [these](https://github.com/Ultimaker/Cura/blob/master/resources/definitions/fdmprinter.def.json) settings have been filled in as local variables for this Python expression.