Ultimaker/Cura #6157: Fails to reach server when multiple instances are open

Liger0, 2019-08-05:
**Application version**
I think all the version since 4.0, I don't know for previous ones

**Platform**
Win10 x64

**Printer**
Any

**Reproduction steps**
Open multiple instances of cura

**Actual results**
All the cura instances after the first report an error about it not being possible to connect to the (ultimaker) server

**Expected results**
I guess it should be able to connect to the server maybe?

**Additional information**

Vandrasc, 2019-08-06:
Hi @Liger0, you mean when you try to log in into your account?

Liger0, 2019-08-06:
No, when Cura starts it instantly prompts the popup "failed to reach the server"

Vandrasc, 2019-08-06:
I am not able to see it, also I cannot image in which situation we will trigger that error, could you please share your log file with us, maybe we can get something from here. Thanks!

Liger0, 2019-08-06:
It seems to happen randomly, not always.
[cura.zip](https://github.com/Ultimaker/Cura/files/3472181/cura.zip)

Vandrasc, 2019-08-07:
Hi @Liger0 , silly question, I see the CloudOutputDeviceManager mentioned in the logs, do you have internet while trying to open the multiple instances? Trying to figure out how this could happen....
Thanks!

Liger0, 2019-08-07:
Yes, I always have internet.

Vandrasc, 2019-08-07:
Could be that you are losing your connection and then it occurs..It is hard to fix something we cannot reproduce. I will have a few more tries but if it is an intermittent bug, I am afraid there is nothing we can do except to keep an eye on it and try to reproduce.

LipuFei, 2019-08-19:
Hi @Liger0, Cura will create a local server with port 32118 to receive callback (sign-in results) from the account service. If you try to sign in at the same time with multiple Cura instances open, I think the problem is that some of the instances will not be able to create a socket to listen on port 32118 because it's occupied by another Cura instance. A workaround is to complete the sign in procedure with one Cura instance at a time. Could you try that?

Liger0, 2019-08-19:
Indeed, I think that's the cause as that works.

LipuFei, 2019-08-19:
@Liger0 So we decided not to fix this at the moment. Sorry for the inconvenience. For now, you can use the workaround to login in one Cura instance at a time. Thanks for reporting this.

closed 2023-06-13 by github-actions[bot]