Ultimaker/Cura #6159: [4.2.1] Right click doesn't work in machines

Liger0, 2019-08-05:
**Application version**
4.2.1

**Platform**
Win10 x64

**Printer**
Crealy cr-10S

**Reproduction steps**
Open the machine edit and go into the gcode fields. Try to right click

**Actual results**
Nothing. Literally.

**Expected results**
It should open a popup to copy, paste, cut, select all

**Additional information**

Vandrasc, 2019-08-06:
Hi @Liger0 , you mean the start/end gcode fields, in the machine settings?

Liger0, 2019-08-06:
Yes

Vandrasc, 2019-08-06:
It seems it stopped working from 4.1 >, will bring it up with the rest of the team.  Nice catch!

LipuFei, 2019-08-07:
We quickly tried adding the context menu for the fields, but with Qt Controls 2, it looks like this by default.
[image]

We don't have a styling for Qt Controls 2 menus yet. After some discussion, we decided not to fix it now. Sorry @Liger0 that you have to use shortcuts for now. Any improvements/PRs will be appreciated!