Ultimaker/Cura #6160: New Cura [4.2.1]: Start gcode is done twice

OlegKor25, 2019-08-06:
Problem: When I tried to print a new stl file for the first time on the new cura 4.2.1, the start gcode would begin, with the nozzle doing a prime line, but right when the line is finished, instead of beginning the part, the nozzle re-homes and does the start gcode again. But after the second time, the start gcode is not repeated and the part starts to print. Thanks for the help!

**Application version**
4.2.1
**Platform**
Windows 10

**Printer**
CR-10s             
Start Gcode:
G28 ; home all axes
G1 Z10 F3000 ; lift
G1 X85 Y0 F1500 ; Jump binder clip
G1 Z0.2 F3000 ; get ready to prime
G92 E0 ; reset extrusion distance
G1 X120 E10 F600 ; prime nozzle
G1 X150 F5000 ; quick wipe

**Reproduction steps**
I tried to reload a file but the start gcode still repeated twice.

**Actual results**
Gcode done twice, but nothing bad happens/printer not damaged.

**Expected results**
The start gcode should be done only once.

fieldOfView, 2019-08-06:
How do you print? Via USB, SD-Card, or OctoPrint?

OlegKor25, 2019-08-06:
SD card. I have not had this issue on 4.0.0 though.

Ghostkeeper, 2019-08-11:
I'm unable to reproduce this issue so far. If I load a small cube into a CR10S printer and save the g-code to a removable drive (USB stick in my case) then I get the following g-code:

```
;FLAVOR:Marlin
;TIME:602
;Filament used: 0.284095m
;Layer height: 0.2
;MINX:133.46
;MINY:133.46
;MINZ:0.2
;MAXX:166.54
;MAXY:166.54
;MAXZ:10
;Generated with Cura_SteamEngine 4.2.1
M140 S50
M105
M190 S50
M104 S200
M105
M109 S200
M82 ;absolute extrusion mode
M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate
M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration
M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

G28 ;Home

G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up

M141 S28
G92 E0
G92 E0
G1 F2700 E-5
;LAYER_COUNT:50
;LAYER:0
M107
G0 F6000 X136.668 Y137.034 Z0.2
;TYPE:SKIRT
G1 F2700 E0
G1 F1200 X137.36 Y136.368 E0.03514
G1 X138.105 Y135.762 E0.07027
G1 X138.898 Y135.22 E0.10542
```

It continues with the skirt from there.

The start g-code is in there only once. Note though that the start g-code specifies that it draws two lines. Maybe that's what you're seeing? That is intended behaviour though.

bjared, 2019-09-05:
I can confirm this on Cura 4.2.1.  I upgraded to this from 3.6.0 (I believe), and it worked properly back then.  This occurred on my first gcode export from 4.2.1.

[image]
[image]

Ghostkeeper, 2019-09-06:
Does one of you have a project file that reproduces the problem?

bjared, 2019-09-06:
Hello,

I created a new project, and exported the gcode, and here is the respective
project and gcode demonstrating this bug.

--Brian

bjared, 2019-09-07:
I think I may have figured it out.  It appears there is now "Extruder" start g-code as well as the normal "Start G-Code".

I started creating a new printer profile, and noticed the content was empty on the extruder settings, and I went back to my existing profile, and it duplicated the start and end g-code to the extruder's settings as well.  Removing the extruder's start/end g-code appears to have removed the duplication!

So, if there's a bug here, I guess it's how Cura handles migrating of the start/end g-code from 3.6 to 4.x?

Ghostkeeper, 2019-09-09:
You attached the project to an e-mail reply. Attachments to e-mail replies to Github don't get published. Probably they get too many e-mail replies with logo images, .ics calendar files and copies of the e-mail in the attachments. Instead of attaching the project file to an e-mail, please visit github.com with a browser and upload it here.

> So, if there's a bug here, I guess it's how Cura handles migrating of the start/end g-code from 3.6 to 4.x?

This setting is not new. It's been there since at least 15.10 but there it seems to have been migrated from the 15.04 code base from when Daid worked on Cura: https://github.com/Ultimaker/Cura/commit/2b81db03ed814e55a2939c81f482614dd093d801 The setting was introduced by Daid there but his version didn't have a version upgrade functionality so the migration wasn't a thing there.

bjared, 2019-09-10:
Sorry about trying to e-mail the zip file.  Here is the project and corresponding gcode it created before I deleted the duplicate entries.

[bug_report.zip](https://github.com/Ultimaker/Cura/files/3593160/bug_report.zip)

Ghostkeeper, 2019-09-10:
The problem here is that the extruder start g-code is executed at the beginning. That didn't used to be the case. That's probably what changed between 3.6 and 4.0. I've heard the report before but I can't find it right now among our open issues.

ninovanhooff, 2019-10-08:
> I've heard the report before but I can't find it right now among our open issues.

@Ghostkeeper sounds like https://jira.ultimaker.com/browse/CURA-6581

maukcc, 2020-01-09:
I would not call this an issue, let alone a bug.
This is wanted behavior. 
1ste start gcode then start extruder gcode, as it is the start of the extruder.

Ghostkeeper, 2020-01-14:
If you want the extruder start g-code in the global start g-code, then you can put `{machine_extruder_start_code, 0}` in the global start g-code. But you need to retain the option to NOT execute the extruder start g-code as part of the global start g-code, because for instance:
* You don't need to wipe the nozzle on a wiping brush at the beginning of the print.
* You needed to heat up the nozzle to execute the start g-code (e.g. print a prime line).
* You need to heat up the nozzle to the Printing Temperature for the global start g-code, but the Initial Printing Temperature after extruder switches.

There are quite a lot of use cases for not wanting to have the extruder's start g-code executed at the beginning of a print. And you have the option of putting the extruder start g-code at the beginning, the end or somewhere halfway your normal start-up sequence.

maukcc, 2020-01-14:
@Ghostkeeper the main flaw in reasoning here, is that it will always start printing with extruder 0.
What happens if you want to start printing with extruder 1 or 5? This is why the start of a print must be seen as an extruder change. We have had this discussion before, and it got changed to the way it is now. Please do not change it back.

fieldOfView, 2020-01-14:
I think it should be possible to do `{machine_extruder_start_code, initial_extruder_nr}`, but I am not sure if the `initial_extruder_nr` is always calculated correctly.

maukcc, 2020-01-14:
@fieldOfView  that would work, but... just tested it. 
{machine_extruder_start_code, initial_extruder_nr} puts in the wrong extruder, and it puts in the literal text: M109 T0 S{material_print_temperature} in stead of M109 T0 S190

fieldOfView, 2020-01-14:
Ah, yes, the extruder start and end gcode snippets would have to be processed *before* the start gcode snippet for this to work.

> {machine_extruder_start_code, initial_extruder_nr} puts in the wrong extruder

Wrong in what way?

maukcc, 2020-01-14:
It always uses extruder 0. Also be sure to switch of brim and skirt as this always starts with extruder 0

maukcc, 2020-01-14:
restarted Cura and now it does something else. The initial_extruder_nr part works. 
But when printing with only extruder 1, it puts in a start 0 and end 0 gcode between start_gcode and start of print.(except when extruder 0 is disabled)

Ghostkeeper, 2020-01-14:
> Ah, yes, the extruder start and end gcode snippets would have to be processed before the start gcode snippet for this to work.

True. I guess that's a bug then.

> It always uses extruder 0. Also be sure to switch of brim and skirt as this always starts with extruder 0, even if extruder 0 is not even used.
Also happens when extruder 0 is disabled

A bug with that was recently found here too: https://github.com/Ultimaker/Cura/pull/6847

rburema, 2020-01-15:
> and it puts in the literal text: M109 T0 S{material_print_temperature} in stead of M109 T0 S190

I tried to reproduce this part of the issue on current master, but for me it kept replacing it correctly. Do you have some profile/project to share where it does happen for you?

rburema, 2020-01-15:
Hang on it _does_ happen on 4.4.1 ... I just can't reproduce it on master from source?

fvrmr, 2021-05-20:
This issue has been removed from our backlog. It is already closed thanks to the no-response bot. 
But our PO and QA-Engineers think we won't work on this anytime soon.

closed 2023-06-13 by github-actions[bot]