Ultimaker/Cura #6161: [4.2.1] Slicing fails and project can't be exported

Liger0, 2019-08-06:
**Application version**
4.2.1

**Platform**
Win10 x64

**Printer**
Cr-10S

**Reproduction steps**
I sliced the model, reduced to 60% in all directions, tried to slice.

**Actual results**
It fails to slice. 
I tried to export the project by "save", "export", and "export selection", but even tho it doesn't prompt any error, it doesn't save anything.

**Expected results**
It should slice it and export the project file.

**Additional information**
[image]
[model.zip](https://github.com/Ultimaker/Cura/files/3472113/model.zip)

nallath, 2019-08-06:
Could you share the log files as well?

Liger0, 2019-08-06:
[cura.zip](https://github.com/Ultimaker/Cura/files/3472160/cura.zip)

nallath, 2019-08-07:
Uh, the logs indicate that the file was written to your removable drive; 
`
2019-08-06 14:18:49,573 - DEBUG - [MainThread] RemovableDriveOutputDevice.RemovableDriveOutputDevice.requestWrite [83]: Writing to E:/CCR10S_Butterfly_stronger-joints.gcode`
Is the file there?

Liger0, 2019-08-07:
Well that's the gcode. I printed that before that error happened. Afterwise I restarted cura I was unable to slice or export the project file.

Ghostkeeper, 2019-08-11:
Since it arrived at the g-code, the slicing at least must've been fine. The log doesn't show that you restarted the application afterwards.

```

2019-08-06 14:18:49,573 - DEBUG - [MainThread] RemovableDriveOutputDevice.RemovableDriveOutputDevice.requestWrite [83]: Writing to E:/CCR10S_Butterfly_stronger-joints.gcode
2019-08-06 14:18:49,616 - INFO - [Thread-12] SliceInfoPlugin.SliceInfoJob.run [29]: Sending anonymous slice info to [https://stats.ultimaker.com/api/cura]...
2019-08-06 14:18:49,959 - INFO - [Thread-12] SliceInfoPlugin.SliceInfoJob.run [33]: Sent anonymous slice info.
2019-08-06 14:18:49,996 - DEBUG - [Thread-1] UM.FileHandler.WriteFileJob.run [71]: Writing file took 0.3750338554382324 seconds
```

It seems to successfully write a file in 0.4 seconds.

I'm also unable to reproduce this problem on my end. It's slicing fine (print takes 1 hour 41 minutes). And it also saves to a project fine ([proof](https://github.com/Ultimaker/Cura/files/3490507/CCR10S_Butterfly_stronger-joints.zip)). This is using 4.2.1 on Windows.

Liger0, 2019-08-14:
As what I remember, I sliced a thing, didn't save, I restarted the PC and resliced and it didn't work, I restarted and it worked again. Maybe I took the log before the latest gcode save, but I think not.. this is the log as of today..
[cura.zip](https://github.com/Ultimaker/Cura/files/3499361/cura.zip)

closed 2024-12-07 by github-actions[bot]