Ultimaker/Cura #6162: Extra loops for raft

jpjokela, 2019-08-06:
Sometimes at the very beginning of the print, there might not be plastic coming off the nozzle as expected. No problem, that's why everyone uses at least skirt, so the problem doesn't affect the model itself.
However, when using raft, this can result in the initial perimeter to not be 100% closed, and if the bed adhesion isn't good enough, doing the raft "infill zigzag" can actually pull the  previous line off rather easily. This doesn't seem to happen much at all, if the zigzag ends in perimeter.

If it was possible to do >1 perimeter, then the first one might be partial, and the second (inner) one should be full.

Granted, this was a LOT bigger issue with my previous print surface, not having major issues anymore, but wouldn't hurt having this as a feature, if anyone else still has similar issues.

rburema, 2019-08-09:
Hi, I understood this as 'skirt for raft'. While this would be/might be a useful addition, we've decided that there are too many other issues/tickets/improvements we need to work on compared to the person-hours we've got available to us. This issue will have to be defferred.

jpjokela, 2019-08-09:
Kind of like that.

Think of replacing the first layer perimeter with skirt in same place, and having the equivalent of "Skirt Line Count" setting available. Extra lines on the outside, and drawn from outside inwards.

And I understand this is not high priority issue. Just thought to share my ideas anyway.

rburema, 2019-08-09:
Sure! You can always at least try :-) 
(Also we welcome pull requests of course ... though even there we have a backlog a.t.m.)

logiclrd, 2020-10-28:
In my personal experience, an incomplete perimeter around the raft hasn't itself caused issues, but I ran into a need for exactly this feature with a very small print, where the nozzle failed to prime before the end of the perimeter and failed to print part of the raft itself. For my use, this would be most useful as a "minimum perimeter line length" parameter, so that it would spiral outward as far as needed to ensure that the requisite distance had been traversed before entering the part. My printer has a long Bowden tube, and in the most recent print I did of this tiny part, I manually added 50mm of additional line prior to it starting the Cura-generated raft bits, and it was actually partway through the real perimeter (!) before filament started coming out. I would probably be setting this value to 55 or 60.

closed 2023-07-13 by github-actions[bot]