Ultimaker/Cura #6163: [4.2.1] In some cases, supports start in the air

jpjokela, 2019-08-06:
Project file showcasing the bug
[bug.zip](https://github.com/Ultimaker/Cura/files/3472543/bug.zip)
**Application version**
4.2.0 & 4.2.1

**Platform**
Linux 64bit

**Printer**
Custom i3 build

**Reproduction steps**
1)Load the attached project
2)Slice with either brim or raft (skirt untested)
3)Observe which layer supports start on
4)With raft, observe lack of air gap between the raft & the model

**Actual results**
Supports start several layers above brim / raft
With raft, there's no air gap between the raft & the model

**Expected results**
Supports start the next layer above brim / raft (as they do on Cura 4.1)
With raft, there's air gap between raft & model

**Additional information**
Loading the same project file to Cura 4.1 seems to result in expected behavior

nallath, 2019-08-07:
Could you check the value of minimum support area?

jpjokela, 2019-08-07:
Looks like I have set it at 0.0. In 4.2.1 I tried 0.5 and 1.0, and supports still start in the air.

jpjokela, 2019-08-07:
It seems that it doesn't happen with bigger layer heights. The object in picture is sliced at 0.04mm layer height. Same seems to happen also with 0.08, but not on 0.12 and higher.

nallath, 2019-08-07:
Could you share your project file? It seems to be a setting related issue, as I'm not getting the same results as you are.

jpjokela, 2019-08-07:
Hm, thought I did, but guess I just saved models and such...

Hope this works?
[bug2.zip](https://github.com/Ultimaker/Cura/files/3477402/bug2.zip)

Ghostkeeper, 2019-08-11:
Yes, that works. That project file initially doesn't reproduce the bug but if I set the Layer Height to 0.04 (down from 0.12) then it does exhibit the problem.

The problem goes away if you set the Support Stair Step Height to 0, indicating that this is a bug with the stair stepping algorithm.

Ghostkeeper, 2019-08-11:
By the way, this is a good bug report. Solid reproduce steps and good analysis. Thanks for that.

jpjokela, 2019-08-12:
Ah, sorry for that. Forgot I was checking if different layer heights behave differently (as I reported, they do) - and simply forgot to switch back to a "bad" value :-)

Thanks for pointing the workaround too, should be able to use latest version again.

Ellecross, 2019-08-13:
hey @jpjokela thanks for closing this issue. But we think it's a valid bug. If you don't mind, I will reopen it. We already created in an internal ticket to track. :) 

For internal cura devs, please see CURA-6715

jpjokela, 2019-08-13:
Didn't mean to do that. Was probably just too tired, and accidentally clicked the wrong "...comment" button below :-)

LipuFei, 2019-08-22:
Fixed via https://github.com/Ultimaker/CuraEngine/pull/1116

closed 2019-08-22 by LipuFei