Ultimaker/Cura #6164: Foever slicing

BagelOrb, 2019-08-06:
**Application version**
4.2.1

**Platform**
Linux

**Printer**
UM S5

**Reproduction steps**
Load project

**Actual results**
Slicing halts at 20%

**Expected results**
Slicing finishes

**Additional information**
I set the infill to 100%, which actually seems to matter. On normal UM s% profile it doesn't halt.

[UMS5_Sample_2__forever_slicing.3mf.renamed.zip](https://github.com/Ultimaker/Cura/files/3472614/UMS5_Sample_2__forever_slicing.3mf.renamed.zip)

[image]

nallath, 2019-08-06:
It also happens for me. There is no crash going on and the engine just keeps using 25% of my CPU.

BagelOrb, 2019-08-09:
Did you actually wait to see if it actually finishes the combing even once?

There might still be an infinite loop inside the combing algorithm!

rburema, 2019-08-09:
@LipuFei said during the CCB where we discussed the ticket that it finishes the 2500 case after about half an hour (if I remember correctly) (I think that might be for one layer??)

LipuFei, 2019-08-09:
@BabelOrb I did. I started the slice and went to a meeting. When I came back (around 2 hours later), I saw that the slice finished.

BagelOrb, 2019-08-09:
Ok good. Then it's just an unfortunately slow slice.

Ghostkeeper, 2019-08-11:
> Because the time complexity of the algorithm is quadratic, this will take ~forever.

It's quartic actually, as in, `O(N^4)`.

BagelOrb, 2019-08-12:
Perhaps this has to do with some code I saw recently which accurately measures the combing travel time when deciding on the print order?

That would certainly cause forever slicing if the combing code is quartic!

smartavionics, 2019-08-12:
Coming late to the party here. It's not actually the combing that is working too hard. It's the wall order optimization that has the problem. With that disabled and the combing enabled it slices in less than 5 seconds.

I will investigate and report back.

BagelOrb, 2019-08-12:
@LipuFei It seems the bug wasn't in combing after all. Take better care in testing next time.

smartavionics, 2019-08-12:
TBH, I cannot yet say with 100% certainty exactly where the problem lies. At the moment it looks like there could be substantial time going into the ordering of the holes which could be related to combing but also there appears to be appreciable further time going into the processing of those holes. On layer 2 of that print, there's 2455 holes and as has been pointed out above, some of the calculations are at least O(n^2). I continue to investigate.

smartavionics, 2019-08-12:
I have discovered where the time is going and will submit one or more PRs in due course. Of course, Cura is always going to struggle with this model given how it handles walls. To satisfy my curiousity, I sliced it using thin walls (my Cura, not Ultimaker's) and the result looks very good. This was printed on a delta in less than 6 minutes. The challenge will be getting it off the blue tape in one piece!

[image]

BagelOrb, 2019-08-13:
That print looks good!

Is this really a single extrusion line?

I would expect more defects in the intersection locations where 3 lines come together.

[IMG_20190812_234559503]<https://user-images.githubusercontent.com/585618/62903822-e8fbcb00-bd5b-11e9-915e-5a3b1f389974.jpg>

smartavionics, 2019-08-13:
> Is this really a single extrusion line?

Yes.

> I would expect more defects in the intersection locations where 3 lines come together.

Looks like this...

[image]

smartavionics, 2019-08-14:
Here's an oddity. If I slice this project with combing turned off (just so that it slices quickly) I get substantially different print time estimates depending on the value of Max Resolution. Using the value in the profile (0.8333) gives a print time of 36 mins but using a much smaller value (0.1) gives a print time of 16 mins. Huh?

[image]

[image]

smartavionics, 2019-08-14:
Oh, cripes! Using a smaller max resolution value actually causes fewer line segments to be output. It's the opposite to what I would expect. I would have thought that using a coarser resolution would cause fine details to be removed but it's the converse that appears to be happening!

smartavionics, 2019-08-14:
Here's a couple more images so you can see the difference changing the max resolution makes...

[image]

[image]

WTF?

BagelOrb, 2019-08-14:
That sounds like a separate issue and it sounds quite important.

There used to be a bug which caused super high poly input to be simplified more than low poly input, but this bug seems to be a different one.

You think it only affects the merging of infill lines for thin walls?

It might even be related to the bug which just got fixed: https://github.com/Ultimaker/Cura/issues/5445

Either way: please make a separate new bug report.

smartavionics, 2019-08-14:
> You think it only affects the merging of infill lines for thin walls?

No, I don't think it's related to thin walls at all. In fact, I have checked that setting the max resolution to 0.1 or 0.8 doesn't make a big change to the output when it is sliced using thin walls so I think the problem is somehow with the normal walls. Maybe the insets are getting mashed up when the resolution is smaller? I don't know.

smartavionics, 2019-08-17:
> Either way: please make a separate new bug report.

Maybe it's related to #6215.

nallath, 2019-12-16:
The pull request for smartavionics fixes this (and just got merged)

closed 2019-12-16 by nallath

BagelOrb, 2019-12-16:
Fixes what?

The max resolution issue, or the slicing time problem?

Closed #6164<https://github.com/Ultimaker/Cura/issues/6164>.

smartavionics, 2019-12-16:
> The max resolution issue, or the slicing time problem?

It fixes the slicing time problem (along with another PR). 

The max resolution issue should be fixed by a pending PR (https://github.com/Ultimaker/CuraEngine/pull/1114). That PR fixes multiple bugs in simplify().

smartavionics, 2019-12-16:
Actually, I'm wrong there. The max resolution issue mentioned above is not fixed by the pending PR. Further investigation required.

smartavionics, 2019-12-16:
I think the walls going away as the resolution is reduced is because the wall widths are so very close to the wall line width setting (0.4 in the profile) that, at times, the walls get removed and that just happens more when the max res value is reduced. If I use a wall line width that is only 10um smaller (i.e. 0.39), all the walls are included even when max res is small (i.e. 0.05). So I think this is a non-issue.

nallath, 2019-12-17:
The forever slicing one. Since that's the title of the issue (and one thread should contain one issue) i closed it.