Ultimaker/Cura #6167: [4.2.1] Number of layers not calculated

Liger0, 2019-08-06:
**Application version**
4.2.1

**Platform**
Win10 x64

**Printer**
Creality cr-10S

**Reproduction steps**
Change the layer height (both the initial and the basic one since the dude that wrote the cre"""awesome""" machine configuration didn't link them).

**Actual results**
The number of layers isn't calculated

**Expected results**
The number of layers should be calculated

**Additional information**
0.84/0.28 == 3 != 7
[image]

Ghostkeeper, 2019-08-07:
I wasn't able to reproduce your case. Maybe that is something that was changed in your custom profile. This is what I did:

1. Choose Super Quality (0.12mm).
2. Change layer height to 0.28mm.
3. Change top/bottom thickness to 0.84mm.

Result:

[image]

However I was able to reproduce something like this with different settings:

1. Choose Standard Quality (0.2mm).
2. Change layer height to 0.3mm.

Top/Bottom Thickness still says 0.8mm. Top layers says 4. 0.8 / 0.3 = 2.67 so even with rounding it shouldn't arrive at 4.

Ghostkeeper, 2019-08-07:
After some investigation I found that this problem is with the GUI not updating the setting values. As per the inheritance formula in the `base_global_standard` profile, the value of Top/Bottom Thickness should be `layer_height_0 + 3 * layer_height`, which would arrive at 1.1mm for my case. However in the interface it stays at 0.8mm.

When I look at the [God Mode](https://github.com/sedwards2009/cura-god-mode-plugin) to see how it arrived at this setting value, I found that the internal setting value is actually set to 1.1mm. If I change the Top/Bottom Thickness to something else and then press reset it does reset to 1.1mm.

The settings being displayed differently is a very serious issue. It could happen for every printer that has inheritance formulas. Maybe it's got something to do with global vs. per-extruder settings?

Vandrasc, 2019-08-09:
Devs, for our internal reference Cura-6708.

nallath, 2019-08-12:
This is actually the result of a major issue in the inheritance formulas. Only those defined in the definition ensure that a value update is triggered. Since the definition defines no relation for top_bottom_thickness, it never gets notified of any updates.

Ghostkeeper, 2019-08-21:
We made a [workaround](https://github.com/Ultimaker/Cura/commit/290b8f4307768aa13b3a788322f945e03faad991) for the next release, but the original issue is still there.

closed 2024-12-07 by github-actions[bot]