Ultimaker/Cura #6171: Update to mint 19.2 with cura 4.2 causing top right of screen to be blank

snavecire, 2019-08-06:
**Application version**

**Platform**

**Printer**

**Reproduction steps**

**Actual results**

**Expected results**

**Additional information**

ewanold, 2019-08-07:
The same for me on Ubuntu 18.04 with versions 4.2, 4.1 and 4.0. 
Cura 3.6 from the Ubuntu repository works fine. 

I suspect one of the last regular updates I did for 18.04 on my machine introduced this issue.

rburema, 2019-08-08:
Do you use the AppImage? (As downloaded from the Ultimaker website.)

ewanold, 2019-08-08:
Sees, all versions are AppImages from Ultimaker. PPA seems to have problems also, so I did not even attempt that. But maybe that is related.

rburema, 2019-08-08:
It's not that, it's just that we support the AppImage (since we need some very specific versions of our dependencies), errors/isues with the PPA's should be brought up with the maintainers.

Since this occurs in the AppImage version as well, we should investigate.

rburema, 2019-08-09:
Could either of you send us the logs? There may be some QML error messages there, if we're lucky.

ewanold, 2019-08-09:
> Could either of you send us the logs? There may be some QML error

"Log" probably means messages on stdout when starting? I already  
checked and it contains nothing useful. Maybe just local minor issues.

Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line  
72: non-double matrix element
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line  
72: non-double matrix element
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf",  
line 80: saw unknown, expected number
qml: TableViewSelection: index out of range
qml: TableViewSelection: index out of range

Running 3.6 from regular Ubuntu yields much more.

/usr/bin/cura
UM.Resources.__initializeStoragePaths [342]: Initializing storage paths
UM.Resources.__initializeStoragePaths [353]: Config storage path is  
/home/ewald/.config/cura/3.1
.....

So I obviously need some changing of a debug level. Unfortunately

./Ultimaker_Cura-4.2.1.AppImage --help

gives me no help. Neither a quick web search. What should I do?

-- 
Ewald Arnold
mailto:ewald at ewald-arnold dot de
mailto:mobil at ewald-arnold dot de

rburema, 2019-08-09:
stdout isn't all of the logs, 'help' -> 'show configuration folder' should show you the folder where the logs are stored.

ewanold, 2019-08-09:
> stdout isn't all of the logs, 'help' -> 'show configuration folder'  
> should show you the folder where the logs are stored.

I removed the existing cura.log, started cura and terminated  
immediately. That minimal file is attatched.

-- 
Ewald Arnold
mailto:ewald at ewald-arnold dot de
mailto:mobil at ewald-arnold dot de

2019-08-09 20:20:20,931 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin FileLogger
2019-08-09 20:20:20,945 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin FirmwareUpdateChecker
2019-08-09 20:20:20,958 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin FirmwareUpdater
2019-08-09 20:20:20,967 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin GCodeGzReader
2019-08-09 20:20:20,977 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin GCodeGzWriter
2019-08-09 20:20:20,986 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin GCodeProfileReader
2019-08-09 20:20:21,012 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin GCodeReader
2019-08-09 20:20:21,025 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin GCodeWriter
2019-08-09 20:20:21,044 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin ImageReader
2019-08-09 20:20:21,054 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin LegacyProfileReader
2019-08-09 20:20:21,068 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin LocalContainerProvider
2019-08-09 20:20:21,086 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin LocalFileOutputDevice
2019-08-09 20:20:21,098 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin MachineSettingsAction
2019-08-09 20:20:21,113 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin ModelChecker
2019-08-09 20:20:21,127 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin MonitorStage
2019-08-09 20:20:21,158 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin PerObjectSettingsTool
2019-08-09 20:20:21,173 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin PostProcessingPlugin
2019-08-09 20:20:21,244 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin PrepareStage
2019-08-09 20:20:21,256 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin PreviewStage
2019-08-09 20:20:21,269 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin RemovableDriveOutputDevice
2019-08-09 20:20:21,303 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SimulationView
2019-08-09 20:20:21,318 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SliceInfoPlugin
2019-08-09 20:20:21,329 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SolidView
2019-08-09 20:20:21,353 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SupportEraser
2019-08-09 20:20:21,408 - DEBUG - [MainThread] cura.OAuth2.AuthorizationService.getAccessToken [112]: No auth data to retrieve the access_token from
2019-08-09 20:20:21,416 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin Toolbox
2019-08-09 20:20:21,434 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin CameraTool
2019-08-09 20:20:21,450 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin MirrorTool
2019-08-09 20:20:21,473 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin RotateTool
2019-08-09 20:20:21,503 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin ScaleTool
2019-08-09 20:20:21,516 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SelectionTool
2019-08-09 20:20:21,538 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin TranslateTool
2019-08-09 20:20:21,570 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin UFPReader
2019-08-09 20:20:21,598 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin UFPWriter
2019-08-09 20:20:21,754 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [372]: Plugin [UM3NetworkPrinting] has been disabled. Skip loading it.
2019-08-09 20:20:21,806 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [372]: Plugin [USBPrinting] has been disabled. Skip loading it.
2019-08-09 20:20:21,821 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin UltimakerMachineActions
2019-08-09 20:20:21,837 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin UpdateChecker
2019-08-09 20:20:21,838 - INFO - [Thread-2] UpdateChecker.UpdateCheckerJob.run [36]: Checking for new version of cura
2019-08-09 20:20:21,870 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade21to22
2019-08-09 20:20:21,885 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade22to24
2019-08-09 20:20:21,900 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade25to26
2019-08-09 20:20:21,914 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade26to27
2019-08-09 20:20:21,928 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade27to30
2019-08-09 20:20:21,941 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin VersionUpgrade30to31
[... cut ...]
2019-08-09 20:22:16,775 - DEBUG - [MainThread] UM.Qt.QtApplication.windowClosed [434]: Shutting down cura
2019-08-09 20:22:16,798 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [292]: Attempting to kill the engine process
2019-08-09 20:22:16,799 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [298]: Killing engine process
2019-08-09 20:22:16,800 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [301]: Engine process is killed. Received return code -15

Ghostkeeper, 2019-08-11:
Attachments don't work when replying to Github notifications via e-mail. Github doesn't post them.

Ghostkeeper, 2019-08-11:
This issue seems to be a duplicate of this one: https://github.com/Ultimaker/Cura/issues/5296

They have a workaround there: Delete the folder `~/.cache/qtshadercache`. That seems to fix the issue for them. Apparently something gets corrupted in Qt's cache files somewhere? Perhaps after a segfault?

closed 2019-08-11 by Ghostkeeper

ewanold, 2019-08-12:
yes, another mystery solved :-)

after removing all the files in the cache directory it works as good  
as before.

-- 
Ewald Arnold
mailto:ewald at ewald-arnold dot de
mailto:mobil at ewald-arnold dot de

snavecire, 2019-08-13:
Sorted,
Thanks to all. 
Deleting the qtshaderscache did the trick.