Ultimaker/Cura #6172: Z Hop elevation not used between test extrusion and print area

sailorgreg, 2019-08-07:
**Application version**
4.2.1

**Platform**
Windows

**Printer**
Creality Ender 3

**Reproduction steps**
Set Z Hop Height to 10. Looking at the GCode for the moves, it adds the Z Hop Height for every other move but not for the initial move to the print area after the test extrusion lines.

**Actual results**
The move between the test extrusion line and the print area happens at the initial print layer elevation.

**Expected results**
I would think that the move would be at the initial print layer elevation + Z Hop Height

**Additional information**
I added the comment 
"; !!! With Z Hop Height set to 10, shouldn't this be 10.3? It keeps running into my previously printed structure !!!" 
on the line I believe is in error.

[CE3_F Letter.gcode.txt](https://github.com/Ultimaker/Cura/files/3476283/CE3_F.Letter.gcode.txt)

Vandrasc, 2019-08-14:
Devs CURA-6718.

LipuFei, 2019-08-21:
Fixed via https://github.com/Ultimaker/CuraEngine/pull/1112

closed 2019-08-21 by LipuFei