Ultimaker/Cura #6173: Cura V2.1 No slicing

alsei12, 2019-08-07:
Hi,
the slicing does not work.
I had the problem already, when I've updated from V3 to V4.1.
The slicing did not work anymore.
I tried a lot ( registry cleaning, cleaning cashes etc.) at the end it worked.

Today I updated V4.1 to V4.2.1. Once again: No slicing.
I work in Windows 10 , I have the Creality CR10 printer.

All the files were working under V3 and V4.1.
Now I have installed the version 4.1 again and the slicing with the same files works well.

[cura.log](https://github.com/Ultimaker/Cura/files/3477131/cura.log)

Vandrasc, 2019-08-07:
Hi @ad1217 , could you save the project file, so we can have your configuration, and share it with us so we can see what the problem is? (File > Save)

Thank you!

alsei12, 2019-08-07:
Hi, thanks for your answer.
Enclosed the project file.

[CCR10_Balkonklemme 03.zip](https://github.com/Ultimaker/Cura/files/3477606/CCR10_Balkonklemme.03.zip)

alsei12, 2019-08-08:
Hi,
it seems that I've fixed the problem.
It was the firewall ( Zonealarm ).
At first I've tried to add the 3 Cura EXE files to the firewall list.
It did not work.
After the complete Zonealarm de-installation, registry cleaning and new installation, Cura 4.2.1 works very well now.

Thanks for your efforts.

closed 2019-08-08 by alsei12