Ultimaker/Cura #6174: Gcode Parsing, Cura 3.6 & 4.2.1

Thisismydigitalself, 2019-08-07:
**Application version**
3.6 and/or 4.2.1

**Platform**
Win10/64

**Printer**
CFFFP

**Reproduction steps**
Load generated Gcode

**Actual results**
 
[image]
[image]

**Expected results**
I was hoping to see the skin as one long uninterrupted line.
Coasting is OFF. print is OK. 

**Additional information**
Why Cura Parser renders the Gcode in such a broken way?

Liger0, 2019-08-09:
It doesn't even work to me. Only 1/2 layers are generated and even in a verìy broken way.

Vandrasc, 2019-08-09:
Even in the compatibility mode I still see a clear and correct generated gcode. Could you please share the gcode with us? Thank you1

Btw, did you did any Windows update lately?

Thisismydigitalself, 2019-08-09:
Ver 4.2.1. i guess this goes the same for 3.6 and others.
I can't share the file in my main post as this belongs to a customer. i do see the same with any stl. i downloaded this rabbit from thingiverse. same outcome. I have attached the gcode which inclused my settings at the bottom of the file. please see if you get the same rendering as mine. I don't believe this got to do with drivers and windows update. slicing workspace looks OK.
[image]

[image]
[image]

[CFFFP_Rabbit.zip](https://github.com/Ultimaker/Cura/files/3485811/CFFFP_Rabbit.zip)

Thisismydigitalself, 2019-08-09:
I just tested this Rabbit model with Cura's built in profiles (not mine) and Gcode parser looks perfect.
I wonder what in my settings is causing this. will try to fiddle with some settings. i wish there was a way to compare profiles. i hope NP++ will help me here and compare between the gcode files settings.

Vandrasc, 2019-08-09:
Oke, that is good to know, at least we now know that it can be one or more of the settings that have been customized and it doesn't happen with the builtin profiles. You can also check very quickly and easily the settings that have been customized by clicking on another profile so the Discard or keep window is triggered. 

Curious what could it be, I was thinking of the Wire printing setting but there you get a message that the preview is not very accurate.

Liger0, 2019-08-09:
[CCR10S_flowCalibration_Cube cura.zip](https://github.com/Ultimaker/Cura/files/3486373/CCR10S_flowCalibration_Cube.cura.zip)
[image]
[FlowCalibration_Cube petg Cura.zip](https://github.com/Ultimaker/Cura/files/3486374/FlowCalibration_Cube.petg.Cura.zip)
[FlowCalibration_Cube petg S3D.zip](https://github.com/Ultimaker/Cura/files/3486375/FlowCalibration_Cube.petg.S3D.zip)

Thisismydigitalself, 2019-08-09:
@Liger0 
Same here with 4.2.1. when I slice using your settings - i get the same gcode. when i slice using Cura's built in profile, model looks perfect. This begs the question - what setting(s) breaks our gcodes?

Thisismydigitalself, 2019-08-09:
Wow, look how beautiful the gcode parser output looks like when using Cura's 0.2 draft profile.
I have NO idea what breaks it in my profile. 

This is not to be confused with Cura's preview workspace - this is the reloaded Gcode:
[image]
I must understand what goes on here.

Thisismydigitalself, 2019-08-09:
Boy!, That was tedious work but I believe I've found the culprit: 
It's **Relative Extrusion** under Special Modes Category.  
this begs another question: does this effects the printed part as it effects the parser?
I will have to print the same part twice to see what/if there is a difference. 
I wonder if the gcode should reflect this setting in such a bad way or this is the correct way to express it.

Thisismydigitalself, 2019-08-09:
BTW, this setting was not included in this Cura message when switching between profiles. i wonder why. this is what Cura prompted to the user when switching between Cura's built in Draft profile and my profile:

[image]

Liger0, 2019-08-09:
It prints fine, even other slicers can read the gcode properly, only cura reads the gcode bad.

Ellecross, 2019-08-14:
Yeah, I can reproduce this too IF the Relative Extrusion setting is TRUE as what @Thisismydigitalself  said. 

And @Liger0 is correct, this is only in the gcode render in Cura. It looks just fine in layer view. 

I'll take this back to the team to discuss.

Ellecross, 2019-08-15:
Hey @Thisismydigitalself. 

We've just discussed this with the team. Unfortunately, we have decided to defer this and will leave this as it is for now. :( But we are now aware that this happens on the Cura's gcode renderer. 

If you see this happening on other settings, please let us know. Thank you for reporting this!

Ghostkeeper, 2019-08-15:
@Ellecross please don't close deferred issues but mark them as deferred instead.

closed 2023-06-13 by github-actions[bot]