Ultimaker/Cura #6175: DANGER ERROR ON CURA 4.x VERSIONS on i3 printers random stops

tmirante, 2019-08-08:
**Application Version**
 4.x all versions

**Platform**
windows 10

**Printer**

no i not put project it happens with all prijnting on Anycubic i3 machine

**Steps to Reproduce**
start printing  after four layer it stops random resiming printing after 20 30 40 50 seconds
**Actual Results**

fter four layer it stops random resiming printing after 20 30 40 50 seconds

**Expected results**
 normal prnting as in cura 3.x versions

**Additional Information**

jpjokela, 2019-08-08:
Not a developer on this project, but it might help if you told some details, such as:
-How do you start the print? USB? SD? Octoprint?
-If it (also) happens from SD, it sounds like the issue would be in the generated gcode, which you probably could share in this bug report? Try if you can replicate it with a simple cube, for example.

tmirante, 2019-08-08:
Yes issue is generated gcode without sense

-------- Messaggio originale --------

tmirante, 2019-08-08:
printer works in stand alone . with 3.x versions all is ok .really is sure a 4.x versin big bug creating me a lot of time spent to find problem and a lot of job/pla lost before understand
it is critical because developers not have answers and i worked on plugin to tune printing platform  at the best.  i can replicate in any moment if you send on printing this file . https://drive.google.com/open?id=1RR9vwIqWon02V--vwK8d0ffMhrZSr8Lx

rburema, 2019-08-09:
You say it's all 4.x versions, the gocode says it's generated with 4.1.0, have you also tested this with 4.2.1?

Because this sounds like a duplicate of [this issue](https://github.com/Ultimaker/Cura/issues/5753) (workarounds are provided [here](https://github.com/Ultimaker/CuraEngine/issues/1096#issuecomment-514109488), but the short of it is that you should take a look at the z-hop/speed settings).

tmirante, 2019-08-10:
No i didn’t but with 3.5.1 is ok . 
I am on job so in those days i have a so not large time to do tests as i lost a lot with 4.1.x . i will do test on september if a recover lost time 

Inviato da Posta per Windows 10

Da: Remco Burema
Inviato: venerdì 9 agosto 2019 14:00
A: Ultimaker/Cura
Cc: tommy; Author
Oggetto: Re: [Ultimaker/Cura] DANGER ERROR ON CURA 4.x VERSIONS on i3 printersrandom stops (#6175)

You say it's all 4.x versions, the gocode says it's generated with 4.1.0, have you also tested this with 4.2.1?
Because this sounds like a duplicate of this issue (workarounds are provided here, but the short of it is that you should take a look at the z-hop/speed settings).

tmirante, 2019-08-21:
let meknowifissueis updatedandthe version to test.things happens inlong printing and my pla isnot infinite. let me know which version is corrected

Ghostkeeper, 2019-08-21:
That issue was fixed in version 4.2.0.

closed 2019-10-31 by Ghostkeeper