Ultimaker/Cura #6179: Display Z height instead/in addition to layer number on print preview

AbeFM, 2019-08-08:
**Is your feature request related to a problem? Please describe.**
Being able to find the location  of a critical feature is one of the best uses of the slicing preview. Often, this input can be critical in determining where to make changes. 

This is also of major concern when layer sizes change for any reason (first layer thickness, variable layer heights).

**Describe the solution you'd like**
[image]
I kept it all the same size.

**Describe alternatives you've considered**
In theory the same information could be gathered by importing the model into an external cad program, then offset by the physical placement of the object in the CURA space. Then do the layer heights math.....

One could create a primitive (support blocker) and scale it until it matches the object... But it still can be hard to get the information.

**Affected users and/or printers**
I think it hits a pretty broad swath of people. I'm kinda afraid it's right there in front of me and I'm missing it. :-)

**Additional context**

[image]

Brand X:
[image]

Brand Y:
[image]

Vandrasc, 2019-08-12:
Devs see CURA-6714, for our internal reference.

DivingDuck, 2019-08-14:
+1 
Would be nice having both information available.

serial8killer, 2019-08-30:
Please do this!!!.....

Ghostkeeper, 2019-09-30:
Duplicate of https://github.com/Ultimaker/Cura/issues/5744

closed 2019-09-30 by Ghostkeeper