Ultimaker/Cura #6180: First nozzle switch not priming the correct distance

BitHappy, 2019-08-09:
**Application version**
Cura 4.2.1

**Platform**
Windows 10

**Printer**
Ender 3 with Dual Extrusion modifications made per this guide: https://docs.google.com/document/d/1WqMb3x0XPMiGRHV5DKVvvKulfL3gMx8mphNqqPGDETM/edit

**Reproduction steps**
I set the "Nozzle Switch Retraction Distance" to 50

**Actual results**
For the first switch between filaments, the first filament retracts 50 as it should, but the second filament fails to push forward 50. Instead it just starts extruding from where it is (50 mm back because of the previous print). I have to watch it until it gets to the first switch and then ram the filament down to the nozzle by hand as fast as possible when the switch is supposed to occur. From then on however all the nozzle switches work fine, retracting 50mm and then priming the other 50mm.

**Expected results**
I expect that it should prime the same distance that the other filament retracted.

**Additional information**
 Or maybe there should be a setting for the initial nozzle switch priming distance or something? In case you change the nozzle switch retraction setting and your machine still has the filament a certain distance back from the previous print? For example, if you have your nozzle switch retraction distance set to 50, then at the end of your print one of the nozzles will be 50mm back. But then if you change the settings to be 70 or something in your next print, you would have to make sure the initial nozzle switch priming distance would still be 50 at first, before carrying on at 70 for both directions for the rest of the print.

I couldn't find any other bug reports on this, but I find it hard to believe I'm the first one to have this issue. Unless maybe there is already a fix for it somewhere and I just don't know about it?Any help would definitely be appreciated! Thanks!

rburema, 2019-08-09:
Just to be sure: Keep in mind that 'Nozzle Switch Retraction Distance' can be set per extruder. If you change the extruder tab in the setting (from 1 to 2), is that still set to 50?

If that's the case, can you please send us the model (or at lest the gcode) you're trying to slice? (Github only accepts zipped files, so you may need to zip or rename before attaching to this ticket.)

BitHappy, 2019-08-12:
Yep all good there! I made sure to change both of them whenever I set the number, and they both work great after the first layer. Its only that first switch where I have to push 50mm in by hand and then it switches back and forth perfectly.

And sure! I have a few different projects I can attach! Mostly just dumb stuff, but hopefully it contains the information you need. Thanks so much!
[VariousMultiMaterialProjects.zip](https://github.com/Ultimaker/Cura/files/3491957/VariousMultiMaterialProjects.zip)

Ghostkeeper, 2019-08-13:
> Or maybe there should be a setting for the initial nozzle switch priming distance or something? In case you change the nozzle switch retraction setting and your machine still has the filament a [...]

Cura in general will always assume that the material is at the tip of the nozzle when the g-code starts. It's possible to change that using the start and end g-code by making it retract at the end of a print and unretract at the beginning, but yes then if you change the retraction length it'll be in the wrong position for the unretract at the beginning.

This goes for both extruders. Cura will assume that your second material is at the nozzle tip when it starts printing for the first time. So it will not unretract by 50mm because it was never retracted by 50mm according to Cura. Cura didn't retract it in any case.

The best way to deal with this in my opinion is to put the retracts and unretracts in the extruders' start and end g-code, something like this:

    G91
    G1 E-50
    G90

And:

    G91
    G1 E50
    G90

AbeFM, 2019-08-13:
Yep, I've been doing a heavy initial purge due to the unpredictability of the last state of the printer - in testing, you often stop mid print with one extruder retracted.

If you HAD to be predictable, unload and reload the filament each print, then you know where you'll be in the tube, I guess. You could put a filament sensor in each tube.... But just prime or do it by hand (something else I do). My retraction is only like 10 mm, so there's less waste in just priming it out.

You have the single-heat-sink switcher thing, right? Can you retract less?

BitHappy, 2019-08-14:
> The best way to deal with this in my opinion is to put the retracts and unretracts in the extruders' start and end g-code, something like this:
> 
> ```
> G91
> G1 E-50
> G90
> ```
> 
> And:
> 
> ```
> G91
> G1 E50
> G90
> ```

Ohh interesting! Awesome! And will that work no matter which extruder you start or end with? Like if you end with extruder 1, but your next print starts with extruder 2, it will move the correct one? Also would that fix the first switch too? Seems like even if it moved the first extruder into position at the beginning of the print, it might still consider the other one to be at the correct spot already like you mentioned, and not work correctly until the second switch. Or am I not understanding completely?
Thanks so much for your replay and help!!

> Yep, I've been doing a heavy initial purge due to the unpredictability of the last state of the printer - in testing, you often stop mid print with one extruder retracted.
>
> If you HAD to be [...]

Yep! I have the NF TC-01. As far as the retract distance goes though, I'm actually not 100% sure how far back I have to retract. The guide I was following said at least 45mm for that particular hotend, so I added an extra 5 just to be safe. Its possible I could retract less though and it might be fine I'll look into that! Thanks!

Ghostkeeper, 2019-08-14:
> Ohh interesting! Awesome! And will that work no matter which extruder you start or end with? Like if you end with extruder 1, but your next print starts with extruder 2, it will move the correct one?

The extruder start and end g-code can be configured per extruder, so you'd want to add that to each of your extruders. It'll be executed with the extruder that you configured the start/end g-code for, so it'll move the correct one yes.

> Also would that fix the first switch too?

No, you'd also need to implement the first switch in the start g-code. Though we had a bug recently that sometimes the extruder start g-code for one of the extruders gets pasted in the start g-code as well, perhaps only if the print starts with a different extruder than the first.

As for an example with the Ultimaker printers:
* In the Ultimaker 3, the material starts in retracted place but the firmware puts the material in the nozzle tip when the prime blob command is output (G280). This then also primes the nozzle in case the material was shifted a bit during downtime.
* In the Ultimaker S5, there is a flow sensor that allows the printer to know exactly when the material is coming out of the nozzle, so it can properly place the material in place when first printing without priming. Cura still puts G280 there but it only indicates to the printer that it needs to move the filament to the tip of the nozzle.
* For both of these printers, the firmware puts the material in a parking position just outside of the heat zone when the `T1` or `T0` command is issued.

closed 2020-06-19 by no-response[bot]