Ultimaker/Cura #6181: Would like to add a new brand of machines to Cura

hussainsail2002, 2019-08-09:
**Is your feature request related to a problem? Please describe.**
I work with a 3D printer manufacturing company in India. We would like to add our machines to the Cura ecosystem of machines. I have created new definition file, extruder file and mesh of my build plate to the appropriate folders under the resources folder. I have added these locally in my system alone, to check whether it works before sending a pull request.

 
I am able to see my machine but when I select it and click next cura generates an error. I am adding the lines under error trace back

Traceback (most recent call last):
  File "X:\4.2\build\inst\lib\python3.5\site-packages\cura\Settings\MachineManager.py", line 405, in addMachine
  File "X:\4.2\build\inst\lib\python3.5\site-packages\cura\Settings\CuraStackBuilder.py", line 68, in createMachine
  File "X:\4.2\build\inst\lib\python3.5\site-packages\cura\Settings\CuraStackBuilder.py", line 133, in createExtruderStackWithDefaultSetup
  File "X:\4.2\build\inst\lib\python3.5\site-packages\cura\Settings\CuraStackBuilder.py", line 129, in createExtruderStackWithDefaultSetup
IndexError: list index out of range

**Describe the solution you'd like**

I am still learning how things work, would be very helpful if anyone can point out the mistake that I am making.

I have added the files to my fork of cura

Cura/resources/definitions/3ding FabxPro.def.json
Cura/resources/extruders/3ding FabxPro_extruder_0.def.json

**Describe alternatives you've considered**

I have not considered any alternatives yet, will do once I figure out what needs to be done

**Affected users and/or printers**
 
My customers 
**Additional context**

fieldOfView, 2019-08-09:
I found your fork here:
https://github.com/Ultimaker/Cura/compare/master...hussainsail2002:master
Next time, please include a link so we don't have to go searching for it.

The issue that is obvious is that the definition does not point to the extruder file:
https://github.com/Ultimaker/Cura/compare/master...hussainsail2002:master#diff-37c91509827c484044b8743e60401a48R15
That should say "3ding FabxPro_extruder_0.def.json"

The platform STL is too big. It needs to be less detailed; the file should be < 1 MB.

hussainsail2002, 2019-08-09:
@fieldOfView 

Thank you, That was it , it now works.

One small issue that I have is that the build platform seems to be positioned elsewhere and not exactly under the print area. Is that also dependent on the file size ?

hussainsail2002, 2019-08-09:
Figured out how to get the bed centered.

It needs to be centered to the global origin on the CAD software before being exported to STL

Thanks @fieldOfView am closing the issue

fieldOfView, 2019-08-09:
You can also use a metadata entry "platform_offset" to shift the platform relative to the origin, eg like here:
https://github.com/Ultimaker/Cura/blob/4.2/resources/definitions/bq_hephestos.def.json#L11

hussainsail2002, 2019-08-12:
@fieldOfView 

Hi sorry to bother you again, I am running into another issue when setting my machines quality, When I set the adhesion_type = Skirt I get the following error 

Traceback (most recent call last):
  File "X:\4.2\build\inst\lib\python3.5\site-packages\cura\BuildVolume.py", line 614, in _onStackChangeTimerFinished
  File "X:\4.2\build\inst\lib\python3.5\site-packages\cura\BuildVolume.py", line 731, in _updateDisallowedAreas
  File "X:\4.2\build\inst\lib\python3.5\site-packages\cura\BuildVolume.py", line 1110, in getEdgeDisallowedSize
  File "X:\4.2\build\inst\lib\python3.5\site-packages\cura\BuildVolume.py", line 1057, in _calculateBedAdhesionSize
Exception: Unknown bed adhesion type. Did you forget to update the build volume calculations for your new bed adhesion type?

If I don't select any adhesion type it automatically sets it to brim, this error only occurs when I select a material from the middle tab.

I tried looking for build volume calculations in the fdmprinter.def.json file but could not locate it, also the material files do not specify any such build volume calculations...

Any help would be most grateful,

Regards,
Hussain

fieldOfView, 2019-08-12:
That should be ```adhesion_type = skirt``` (lower case)

hussainsail2002, 2019-08-12:
Yess !!

That worked, Thank you !!