Ultimaker/Cura #6184: Cura error every time It starts

thomllama, 2019-08-09:
Cura wont start simply, I get this error every time.  done everything I can think of including deleting older versions wiping all ultimaker/cura files from the computer, downloading fresh Cura installer.. nothing 
[image]

nallath, 2019-08-12:
If you don't copy the actual detailed crash report, I don't know which one you had. We mean it when we make the crash reports anonymous.

fieldOfView, 2019-08-12:
Seems to have been resolved here:
https://community.ultimaker.com/topic/29010-ultimaker-cura-42-released/?do=findComment&comment=244856

LipuFei, 2019-08-14:
Thanks @fieldOfView ! I'll close this issue.

closed 2019-08-14 by LipuFei