Ultimaker/Cura #6185: [4.2.1] Supports not touching ground

TixoRebel, 2019-08-09:
**Application version**
4.2.1

**Platform**
Windows 10 (Multiple patches)

**Printer**
CR-10S Pro
[CCR10SPRO_Support_Test.zip](https://github.com/Ultimaker/Cura/files/3486656/CCR10SPRO_Support_Test.zip)

**Reproduction steps**
1. Setup CR-10S Pro printer
2. Select super quality
3. Enable supports
4. Set support placement to everywhere
5. Open [this model](https://www.shapeways.com/product/5TSTBEFSE/rug-of-smothering)
6. Slice model

Profile and model also included in the zip above.

**Actual results**
Some supports do not generate on layers 1-3.

**Expected results**
Supports should generate on layers 1-3.

**Additional information**
I have been able to solve most of the other issues regarding generating supports for this model by tweaking settings such as changing minimum support area to 0 or by enabling towers. However, I have been unable to get it to generate all the supports on the first 3 layers (ends up being the first 5 layers if the layer height is dropped to 0.08mm). I have also seen it generate parts of supports on the first 3 layers but not others in the same slice.

**Layers 1-3**
[image]

**Layer 4**
[image]

rburema, 2019-08-13:
Confirmed. As a workaround, set the Support Stair Step height to less than the layer-height, and the first layer (and onwards) will show up.

Ghostkeeper, 2019-08-13:
Duplicate of https://github.com/Ultimaker/Cura/issues/6163.

closed 2019-08-13 by Ghostkeeper

Schleichmichl, 2019-08-16:
Same problem. Unfortunately, reducing stair step height doesn't have any affect.
When fixing this, please check also if support brim is generated when enabled.
Added stl file to reproduce

[image]

[RightTop.zip](https://github.com/Ultimaker/Cura/files/3508622/RightTop.zip)

rburema, 2019-08-16:
@Schleichmichl We're (also) aware of the support-brim issue, the fix for that is in progress (by coincindence, I worked on it just an hour or so ago).