Ultimaker/Cura #6186: Make the model preview show correctly

Liger0, 2019-08-09:
**Is your feature request related to a problem? Please describe.**
Yes, I can't see the models correctly and can't move, rotate etc correctly as I can't see them as I would like to.

**Describe the solution you'd like**
Change color instead of that dark gray, and make the shades more visible

**Describe alternatives you've considered**
-

**Affected users and/or printers**
Cr-10s but probably all of them

**Additional context**
[image]

nallath, 2019-08-13:
If you rotate your camera, do you see the shading change? Because that is the effect i'm getting (eg; The camera has a light source on the same position, which should make specular reflection be visible on the model)

fieldOfView, 2019-08-13:
If the 3d view had less ambient light, more shading would be visible on darker colors.

Ghostkeeper, 2019-08-13:
No, reducing the ambient light has no effect. It only makes the object darker. We do have diffuse (Lambert) shading, but since the light is in the camera there will always be a position where multiple different normal vectors are rendered with the same colour. I can do the same thing with this calibration model:

[image]

The real problem is that the light source is always located on the camera. Increasing the strength of diffuse shading relative to ambient makes curves more pronounced, but it won't force some surfaces to become darker than others because they happen to be oriented away from the light; the light can always be moved so that up to 3 surfaces are equally far rotated.

However placing the light in a fixed location makes it hard to see things that then fall into the shadow.

fieldOfView, 2019-08-13:
Quite right.

How about two lights then? One on the camera, one fixed position.

Ghostkeeper, 2019-08-14:
That could be a way to approach this, yes. The fixed light shouldn't need to be very strong.

LipuFei, 2019-08-15:
For dev, JIRA ticket: CURA-6723

Liger0, 2020-02-27:
Still not solved. :\

nallath, 2022-06-15:
We've deferred this issue in the meantime. So no.

FireflyJar, 2023-02-15:
I am new to Cura and was having same issue as Liger0  but figured it out!
I had my material set to Black, I thought the material  color is for color of filament and did not know it was going to change color of model on display.  Simplify3d did not do this so  I did not expect it.

CWempe, 2023-06-06:
I also think this should be fixed.
There is just no reason the object should be shown as a "matte black 2D object" like this.
[image]

As a workaround it looks like you could define black as `#111111` instead of `#000000`.
[image]

This seems to help a little and still is presented as "black".
[image]

Maybe Cury could just make `#000000` less black internaly since the object is shown in dark gray anyway.

rkg82tg, 2024-08-21:
A bit more than a year later since the last post in this thread ...
Any chance of seeing an improvement in lighting in the Prepare view this year? 
Although the current workaround of changing the material color to a dark shade of grey improves visibility, haveing a second light source would be great, or just being able to change the position of the single light source.
Thanks for considering.

GregValiant, 2024-11-22:
I'm cleaning house.
Is this still a problem with current Cura versions (5.8.0 and up)?  Can I close this?

rkg82tg, 2024-11-22:
The problem appears to have been resolved in v. 5.9.0, so I have nothing personally against closing this ticket.

closed 2024-11-22 by GregValiant