Ultimaker/Cura #6187: Maximum resolution, deviation documented wrong(?)/insufficiently

richfelker, 2019-08-09:
The units for `meshfix_maximum_resolution` and `meshfix_maximum_deviation` are mm according to their descriptions in `fdmprinter.def.json`; however, based on `utils/polygon.cpp` in CuraEngine, they seem to be used in comparisons against quantities with units of mm². I can't find any documentation for what exactly they mean or how to set them correctly, nor on which takes precedence if they conflict (i.e. if preventing excess deviation would require violating the minimum segment size).

rburema, 2019-08-13:
Take a look at lines 903 and 904 of polygon.h (of Polygon**s**), in the simplify specified there:

~~~
const coord_t allowed_error_distance_squared = allowed_error_distance * allowed_error_distance;
const coord_t smallest_line_segment_squared = smallest_line_segment * smallest_line_segment;
~~~

From there on it, calls the simplify-method (of PolygonRef) in the cpp file, with those values, and thus with the correct units.

Ghostkeeper, 2019-08-13:
Documentation is here: https://github.com/Ultimaker/CuraEngine/blob/e372570f048c449e4083b23f7f0c3fc25812938d/src/utils/polygon.h#L884-L900

I've made a small improvement in the user-visible documentation as well. I think that suffices for this ticket.

commit 66939192 (2019-08-13): Expand documentation of Maximum Deviation

Fixes #6187.
closed 2019-08-13 by Ghostkeeper