Ultimaker/Cura #6188: [4.1.0] Adaptive Layer & Support Z Distance

silycr, 2019-08-10:
**Platform**
Cura 4.1.0
Windows 10

**Printer**
TronXY X5S-2e

Ghostkeeper, 2019-08-13:
Indeed, CuraEngine makes these calculations by dividing the desired distance by the layer height. The Adaptive Layer Height was implemented naively and the most difficult part of it to get this to work was never tackled, which is why it's still in Experimental.

We have a pull request that tries to address this for the top/bottom thickness here: https://github.com/Ultimaker/CuraEngine/pull/1094/files I have yet to take a look at that PR at all, but there is an inherent conflict in the front-end settings of thickness vs. number of layers. For support Z distance this can be fixed but for other things I don't think there is a good solution.

silycr, 2019-08-18:
Thanks for the confirmation. I'm not too familiar with the CuraEngine calculations. 
It's a bit unfortunate as I enjoy using adaptive layers. Quite useful to be able to print angled walls fast at a fine setting and use the overhead melt flow to print thicker layers on vertical walls. 

Currently just tweaking z offset provides a workaround, generally assuming reduced layer height at interfaces.