Ultimaker/Cura #6189: Wiki: SDK overview needs an update

thopiekar, 2019-08-10:
Hey!

Just came to this wiki page and noticed that there is no update since Cura 4.0 (SDK 4.0.0).
Could you take some time and give it an update?

Thanks!

Ghostkeeper, 2019-08-13:
@diegopradogesto Could you dig through your sent e-mail to see what update notes you sent out?

ChrisTerBeke, 2019-08-13:
The following notes are stored in the contributor portal:

**SDK 6.0.0 (Cura 4.0.0)**
* Major UI redesign
* New Login functionality with Ultimaker Account
* New settings: 'Minimum support area' and 'minimum support interface'

**SDK 6.1.0 (Cura 4.1.0)**
* Printer discovery: in order to make printers discoverable over the network, plugins can use the DiscoveredPrintersModel. The plugin finds a bunch of printers, user can select a printer and then the plugin can register the printer in Cura and make it active.
* First run wizard: the flow when first time users run Ultimaker Cura has changed. Now it shows a wizard-like dialog showing some pages one after the other. If a plugin wants to make use of it, the best option is to create a model based on the WelcomePagesModel (as an example, take a look at AddPrinterPagesModel).

**SDK 6.2.0 (Cura 4.2.0)**
* Orthographic view. A new way to visualise the models in the build plate. Previously only the perspective view was possible but from this version, the orthogonal view can be selected in the "View > Camera View" menu.
* Object list. At the left bottom of the screen, now a list of the models that are on the build plate shows up. There is a new model (ObjectsModel) which contains that information. New plugins can use this model to access this data.
* Flow per feature. New settings were added to control the flow of different parts of the prints, such as infill, walls, skin, ... The print profiles can now make use of these settings.

I'm not sure what information we want to add to the Wiki, but it's a start.

Ghostkeeper, 2019-08-14:
No, that's precisely what I meant. It's what we've written for the previous SDK updates too. I'll move that to the wiki.

Ghostkeeper, 2019-08-14:
I've updated it now. Thanks.

closed 2019-08-14 by Ghostkeeper

thopiekar, 2019-08-14:
Thanks @Ghostkeeper !