Ultimaker/Cura #6190: Holes in slicing

techfreak111, 2019-08-10:
**3.5.0**

**Windows**
Nvidia Quadro M500M, Nvidia Quadro M550

**Anycubic I3 Mega**

**2 different Computers, the same Version, 2 different slices**

**On 1 Computer everything is fine, on the other i have the holes above und underneath 1 letter. I have tried an update on the pc with the holes, but on the actual Version of cura, i think something with 4. i had the same failure with the holes**

**what changes i may make in the Options to have not the holes in the slicing?**

[image]

[image]

Vandrasc, 2019-08-11:
Hi @techfreak111 , could you please share your project file with us so we a have a look at your configurations and what settings you have customized. Use the same configuration as in the one where you encounter the holes in the model then go to File>Save...

Thank you!

rburema, 2019-08-13:
If it's the exact same file, with the exact same settings, it _might_ be a visual artifact. Are the drivers of your video card up to date on both machines?

Ghostkeeper, 2019-08-13:
Looks like Coasting to me. Did you accidentally enable coasting?

Liger0, 2019-08-18:
I also had that weird issue on cura 3.6, and there wasn't coasting.

closed 2020-06-19 by no-response[bot]